MrBuddyCasino/ESP32_Alexa #1: compile error

jhpark555, 2017-08-02:
There is an compile error like below. I didn't change anything.

esp32_alexa/components/asio/asio_http.c:18:19: fatal error: brssl.h: No such file or directory
compilation terminated.
make[1]: *** [/c/esp-idf/make/component_wrapper.mk:211: asio_http.o] Error 1
make: *** [C:/esp-idf/make/project.mk:386: asio-build] Error 2

MrBuddyCasino, 2017-08-02:
Did you do `git submodule init && git submodule update`?

jhpark555, 2017-08-02:
Yes. I did it but the result is same.  So I changed the code like this.  There are some places to be changed like this.
 #include "../bear_ssl_tools/brssl.h"

Then those errors are gone but another errors happen like below.

D:/esp32/esp32_alexa/main/app_main.c:116:13: warning: 'start_web_radio' defined but not used [-Wunused-function]
 static void start_web_radio()
             ^
D:/esp32/esp32_alexa/main/app_main.c:149:13: warning: 'signal_strength' defined but not used [-Wunused-function]
 static void signal_strength()
             ^
AR libmain.a
LD alexa.elf
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.cc_do_sign+0x18): undefined reference to `br_rsa_pkcs1_sign_get_default'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.cc_do_sign+0x1c): undefined reference to `br_ecdsa_sign_asn1_get_default'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.cc_do_sign+0x20): undefined reference to `br_ec_get_default'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.run_command+0x2c): undefined reference to `br_ssl_engine_close'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.run_command+0x30): undefined reference to `br_ssl_engine_renegotiate'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x1c): undefined reference to `br_tls10_prf'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x20): undefined reference to `br_tls12_sha256_prf'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x24): undefined reference to `br_tls12_sha384_prf'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x30): undefined reference to `br_ssl_client_zero'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x34): undefined reference to `br_x509_minimal_init'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x38): undefined reference to `br_ssl_engine_set_default_aes_cbc'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x3c): undefined reference to `br_ssl_engine_set_default_aes_gcm'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x40): undefined reference to `br_ssl_engine_set_default_chapol'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x44): undefined reference to `br_ssl_engine_set_default_des_cbc'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x48): undefined reference to `br_ssl_client_set_default_rsapub'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x4c): undefined reference to `br_ssl_engine_set_default_ec'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x50): undefined reference to `br_ssl_engine_set_default_rsavrfy'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x54): undefined reference to `br_ssl_engine_set_default_ecdsa'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x58): undefined reference to `br_ssl_engine_set_suites'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x5c): undefined reference to `br_rsa_pkcs1_vrfy_get_default'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x60): undefined reference to `br_ecdsa_vrfy_asn1_get_default'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x64): undefined reference to `br_ssl_engine_set_buffer'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_connect+0x68): undefined reference to `br_ssl_client_reset'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_run_engine+0x2c): undefined reference to `br_ssl_engine_current_state'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_run_engine+0x30): undefined reference to `br_ssl_engine_recvapp_buf'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_run_engine+0x34): undefined reference to `br_ssl_engine_recvapp_ack'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_run_engine+0x38): undefined reference to `br_ssl_engine_sendrec_buf'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_run_engine+0x3c): undefined reference to `br_ssl_engine_sendrec_ack'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_run_engine+0x40): undefined reference to `br_ssl_engine_recvrec_buf'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_run_engine+0x44): undefined reference to `br_ssl_engine_recvrec_ack'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_run_engine+0x48): undefined reference to `br_ssl_engine_sendapp_buf'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_run_engine+0x4c): undefined reference to `br_ssl_engine_sendapp_ack'
D:/esp32/esp32_alexa/build/asio\libasio.a(asio_secure_socket.o):(.literal.asio_ssl_run_engine+0x50): undefined reference to `br_ssl_engine_flush'
[... cut ...]
D:/esp32/esp32_alexa/build/bear_ssl_tools\libbear_ssl_tools.a(names.o):(.rodata.hash_functions+0x28): undefined reference to `br_sha256_vtable'
D:/esp32/esp32_alexa/build/bear_ssl_tools\libbear_ssl_tools.a(names.o):(.rodata.hash_functions+0x34): undefined reference to `br_sha384_vtable'
D:/esp32/esp32_alexa/build/bear_ssl_tools\libbear_ssl_tools.a(names.o):(.rodata.hash_functions+0x40): undefined reference to `br_sha512_vtable'
collect2.exe: error: ld returned 1 exit status
make: *** [C:/esp-idf/make/project.mk:323: /d/esp32/esp32_alexa/build/alexa.elf] Error 1

MrBuddyCasino, 2017-08-02:
Thanks for the bug report. I'll try to check out fresh and re-build to see if I can reproduce the issue. At first glance, it looks like bearssl is missing. I have an idea whats going on.

Do you have components/bear_ssl in your project? Is it empty?
If it isn't, try to create a file "component.mk" in that directory with this content:

```
COMPONENT_ADD_INCLUDEDIRS := .	\
							inc \
							src \
							tools
							
COMPONENT_SRCDIRS +=	src/codec \
						src/ec \
						src/hash \
						src/int \
						src/mac \
						src/rand \
						src/rsa \
						src/ssl \
						src/symcipher \
						src/x509 \
						src 
CFLAGS += -DBR_USE_ALT_RAND -DBR_USE_UNIX_TIME```

This should add bearssl to the project components.

jhpark555, 2017-08-02:
I appreciate that. It works fine. The component.mk missing was a problem. 
Thanks

ESP32DE, 2017-08-02:
@MrBuddyCasino 
after 6 minutes.. ( make all ) 
> .....
> ....
> make[1]: Leaving directory 'C:/Neon3/buddy_Alexa1/build/xtensa-debug-module'
> Target 'component-xtensa-debug-module-build' responsible for 'C:/Neon3/buddy_Alexa1/build/xtensa-debug-module/libxtensa-debug-module.a'
> xtensa-esp32-elf-gcc -nostdlib -u call_user_start_cpu0  -Wl,--gc-sections -Wl,-static -Wl,--start-group  -LC:/Neon3/buddy_Alexa1/build/MerusAudio -lMerusAudio -LC:/Neon3/buddy_Alexa1/build/alexa -lalexa -LC:/Neon3/buddy_Alexa1/build/app_trace -lapp_trace  -LC:/Neon3/buddy_Alexa1/build/app_update -lapp_update -LC:/Neon3/buddy_Alexa1/build/asio -lasio -LC:/Neon3/buddy_Alexa1/build/audio_player -laudio_player -LC:/Neon3/buddy_Alexa1/build/audio_recorder -laudio_recorder -LC:/Neon3/buddy_Alexa1/build/audio_renderer -laudio_renderer -LC:/Neon3/buddy_Alexa1/build/aws_iot  -LC:/Neon3/buddy_Alexa1/build/bear_ssl -lbear_ssl -LC:/Neon3/buddy_Alexa1/build/bear_ssl_tools -lbear_ssl_tools -LC:/Neon3/buddy_Alexa1/build/bootloader_support -lbootloader_support -LC:/Neon3/buddy_Alexa1/build/bt -lbt -LC:/Neon3/buddy_Alexa1/build/bt_speaker  -LC:/Neon3/buddy_Alexa1/build/coap -lcoap -LC:/Neon3/buddy_Alexa1/build/common -lcommon -LC:/Neon3/buddy_Alexa1/build/controls -lcontrols -LC:/Neon3/buddy_Alexa1/build/cxx -lcxx -u __cxa_guard_dummy -LC:/Neon3/buddy_Alexa1/build/driver -ldriver -LC:/Neon3/buddy_Alexa1/build/esp32 -lesp32 C:/esp32neon3/esp-idf-3126/components/esp32/libhal.a -LC:/esp32neon3/esp-idf-3126/components/esp32/lib -lcore -lrtc -lphy -lcoexist -lnet80211 -lpp -lwpa -lsmartconfig -lcoexist -lwps -lwpa2 -L C:/esp32neon3/esp-idf-3126/components/esp32/ld -T esp32_out.ld -u ld_include_panic_highint_hdl -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -LC:/Neon3/buddy_Alexa1/build/ethernet -lethernet -LC:/Neon3/buddy_Alexa1/build/expat -lexpat -LC:/Neon3/buddy_Alexa1/build/fatfs -lfatfs -LC:/Neon3/buddy_Alexa1/build/fdk-aac -lfdk-aac -LC:/Neon3/buddy_Alexa1/build/fdk-aac_decoder -lfdk-aac_decoder -LC:/Neon3/buddy_Alexa1/build/fifo -lfifo -LC:/Neon3/buddy_Alexa1/build/freertos -lfreertos -Wl,--undefined=uxTopUsedPriority -LC:/Neon3/buddy_Alexa1/build/heap -lheap -LC:/Neon3/buddy_Alexa1/build/http -lhttp -LC:/Neon3/buddy_Alexa1/build/jsmn -ljsmn -LC:/Neon3/buddy_Alexa1/build/json -ljson -LC:/Neon3/buddy_Alexa1/build/libfaad -llibfaad -LC:/Neon3/buddy_Alexa1/build/libfaad_decoder -llibfaad_decoder -LC:/Neon3/buddy_Alexa1/build/libm4a -llibm4a -LC:/Neon3/buddy_Alexa1/build/libsodium -llibsodium -LC:/Neon3/buddy_Alexa1/build/log -llog -LC:/Neon3/buddy_Alexa1/build/lwip -llwip -LC:/Neon3/buddy_Alexa1/build/mad -lmad -LC:/Neon3/buddy_Alexa1/build/main -lmain -LC:/Neon3/buddy_Alexa1/build/mbedtls -lmbedtls -LC:/Neon3/buddy_Alexa1/build/mdns -lmdns -LC:/Neon3/buddy_Alexa1/build/mdns_task -lmdns_task -LC:/Neon3/buddy_Alexa1/build/micro-ecc -lmicro-ecc -LC:/Neon3/buddy_Alexa1/build/mp3_decoder -lmp3_decoder -LC:/Neon3/buddy_Alexa1/build/multipart_parser -lmultipart_parser -LC:/Neon3/buddy_Alexa1/build/newlib -lnewlib C:/esp32neon3/esp-idf-3126/components/newlib/lib/libc.a C:/esp32neon3/esp-idf-3126/components/newlib/lib/libm.a -LC:/Neon3/buddy_Alexa1/build/nghttp -lnghttp -LC:/Neon3/buddy_Alexa1/build/nghttp_client -lnghttp_client -LC:/Neon3/buddy_Alexa1/build/nvs_flash -lnvs_flash -LC:/Neon3/buddy_Alexa1/build/openssl -lopenssl -LC:/Neon3/buddy_Alexa1/build/sdmmc -lsdmmc -LC:/Neon3/buddy_Alexa1/build/servo -lservo -LC:/Neon3/buddy_Alexa1/build/sntp -lsntp -LC:/Neon3/buddy_Alexa1/build/soc -lsoc -LC:/Neon3/buddy_Alexa1/build/sounds -lsounds -LC:/Neon3/buddy_Alexa1/build/spi_flash -lspi_flash -LC:/Neon3/buddy_Alexa1/build/tcpip_adapter -ltcpip_adapter -LC:/Neon3/buddy_Alexa1/build/ui -lui -LC:/Neon3/buddy_Alexa1/build/ulp -lulp -LC:/Neon3/buddy_Alexa1/build/url_parser -lurl_parser -LC:/Neon3/buddy_Alexa1/build/user_driver -luser_driver -LC:/Neon3/buddy_Alexa1/build/vfs -lvfs -LC:/Neon3/buddy_Alexa1/build/wear_levelling -lwear_levelling -LC:/Neon3/buddy_Alexa1/build/web_radio -lweb_radio -LC:/Neon3/buddy_Alexa1/build/wifi -lwifi -LC:/Neon3/buddy_Alexa1/build/wpa_supplicant -lwpa_supplicant -LC:/Neon3/buddy_Alexa1/build/xtensa-debug-module -lxtensa-debug-module -lgcc -lstdc++ -Wl,--end-group -Wl,-EL -o C:/Neon3/buddy_Alexa1/build/alexa.elf -Wl,-Map=C:/Neon3/buddy_Alexa1/build/alexa.map
> python C:/esp32neon3/esp-idf-3126/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode "dio" --flash_freq "40m" --flash_size "16MB"  -o /c/Neon3/buddy_Alexa1/build/alexa.bin C:/Neon3/buddy_Alexa1/build/alexa.elf
> esptool.py v2.1-beta1
> To flash all build output, run 'make flash' or:
> python C:/esp32neon3/esp-idf-3126/components/esptool_py/esptool/esptool.py --chip esp32 --port COM27 --baud 921600 --before default_reset --after hard_reset write_flash -u --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 C:/Neon3/buddy_Alexa1/build/bootloader/bootloader.bin 0x10000 /c/Neon3/buddy_Alexa1/build/alexa.bin 0x8000 C:/Neon3/buddy_Alexa1/build/partitions.bin
> 
> 20:25:55 Build Finished (took 6m:57s.931ms)
> 
> 

confirm the component.mk solve this ( inc - > include <-> inc ) too
well done!

best wishes
rudi ;-)

MrBuddyCasino, 2017-08-03:
Thanks for testing this. Don't know how to fix this yet.
Btw. you can speed up make by passing it the "-j4" parameter to use 4 cores in parallel.

MrBuddyCasino, 2017-08-11:
I changed the way the submodule is included, it should now work out of the box. Closing this issue, please make a new one should there still be issues.

closed 2017-08-11 by MrBuddyCasino