MrBuddyCasino/ESP32_Alexa #3: What's the current status of this code?

martinbradford, 2017-08-10:
I've pulled down the most recent commit and got it to compile - with the observation that I had to edit "freertos.h" to "FreeRTOS.h" to get a clean compilation under Unix - case sensitive! I believe that I have completed all the configuration and build steps and it flashes cleanly, but does not run successfully. When I view the output on the monitor, I get the following : 

I (6074) nghttp2: Writing HTTP request...
I (6074) nghttp2: Request headers:
I (6074) nghttp2: :method: GET
I (6084) nghttp2: :scheme: https
I (6084) nghttp2: :authority: alexa.boeckling.net
I (6094) nghttp2: :path: /auth/refresh/Atzr|IwEBIMuV5r1sH7IglsMkC4ZY4GQXEAgbbKRzfZwgMVhAW6v41vzc9_8nE_EYCMO1zM44hI4i-Kx78bTq6jkZz9LX9rYcZHniQtpvVjgDuQhRZSQfzeQycMk5NEMbjoGmOa3SJElWU9LHECI7UVTe3G-xbwzA8FucwtPM6x1E__BM0loa-dFjKa7mwVG2sQbbOO2mKhfjx9oF9MZf4hZcR6eHF2JzNXKKTx5PN9UfZUhX1HhhjTtjaFEISSfDODa93pNzGd-kiPODFPxcPOqVhsXXfwb0bv5ixnHMStzpMq73GaVW1h-QaQGPX7iueXjV0OWFfjDkkjvW8XcAs2UFHo_id5AVnZSO2Acs146H0bLJCuYEjdJm0_qnubMDDEVY3YFhch2beXEose5ojN1u3sAoVFkKSCxASsRvfApuRDbKpnxVLtDiNcSERakDoC4YJUjTp71aXe7FreXkBEoUPpgSaricAps_v0ftv0tgMv7RnonLmVJk7K9wyq1Py1BGMxmKVKkQB6ZXylqIWSk_1o936bLguCEUc6xzF3w4cfwKLsaVIAXe7n-juxFipBk17prmo9k

I (6144) nghttp2: created new stream: 1
W (6154) nghttp2: 859: - RAM left 141824
I (6154) asio: adding task: alexa.boeckling.net
Algorithms:
I (6164) asio: removing task: refresh_auth_token
W (6164) asio: 96: - RAM left 134244
connecting to: 188.226.169.151
connected.
WARNING: no configured trust anchor
W (6254) led_ui: switching anim
E (6254) led_ui: setColors 0=5, 1=16389
ERROR: SSL error 8 (BR_ERR_NO_RANDOM)
I (6254) asio_handler_ssl: asio_ssl_handle_close
I (6254) nghttp2: freeing http2_session_data
W (6264) nghttp2: 194: - RAM left 161152
W (6264) asio: 96: - RAM left 161212
I (13564) wifi: pm start, type:0

Then I hit the GPIO0 button and it simply reboots with the following traceback : 

GPIO[0] intr, val: 1
Guru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
Register dump:
PC      : 0x400014fd  PS      : 0x00060930  A0      : 0x801173d2  A1      : 0x3ffce190  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffce1b0  
A10     : 0x00000003  A11     : 0x00060923  A12     : 0x00060920  A13     : 0x00000080  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fd:0x3ffce190 0x401173cf:0x3ffce1a0 0x40117621:0x3ffce1c0 0x401178ee:0x3ffce220 0x40117416:0x3ffce240 0x401189b1:0x3ffce260 0x401189d7:0x3ffce290 0x40118577:0x3ffce2b0 0x401185e5:0x3ffce2d0 0x4011773e:0x3ffce2f0 0x401007b3:0x3ffce320
0x401173cf: build_auth_header at /home/martin/esp/ESP32_Alexa/components/alexa/./alexa.c:499

0x40117621: alexa_send_event at /home/martin/esp/ESP32_Alexa/components/alexa/./alexa.c:499

0x401178ee: speech_recognizer_start_capture at /home/martin/esp/ESP32_Alexa/components/alexa/./alexa_speech_recognizer.c:169

0x40117416: alexa_gpio_handler at /home/martin/esp/ESP32_Alexa/components/alexa/./alexa.c:499

0x401189b1: asio_gpio_run at /home/martin/esp/ESP32_Alexa/components/asio/./asio_gpio.c:96

0x401189d7: asio_gpio_event at /home/martin/esp/ESP32_Alexa/components/asio/./asio_gpio.c:111

0x40118577: asio_registry_poll_connection at /home/martin/esp/ESP32_Alexa/components/asio/./asio.c:105

0x401185e5: asio_registry_poll at /home/martin/esp/ESP32_Alexa/components/asio/./asio.c:152

0x4011773e: alexa_init at /home/martin/esp/ESP32_Alexa/components/alexa/./alexa.c:605 (discriminator 1)

0x401007b3: alexa_task at /home/martin/esp/ESP32_Alexa/main/./app_main.c:51

Rebooting...

I do appreciate that this is still code in development - just want to avoid wasting too much time code reading and debugging if there is either a simple fix, or the answer is that it is still too experimental to be worth my time following it through!

Martin

MrBuddyCasino, 2017-08-10:
`ERROR: SSL error 8 (BR_ERR_NO_RANDOM)`

This looks like a problem. The BearSSL submodule might have changed, not sure. Someone else had this issue as well I think, I'll try to look into it.

Can you make a pull request with the capitalization fixes? I'm on a Mac and can't easily test that.

martinbradford, 2017-08-10:
I shall try - I'm not very familiar with this GitHub thing - I use CollabNet for my cloud development platform, so this may not work out!

MrBuddyCasino, 2017-08-10:
It looks complicated at first, but its really just 
- fork
- push changes
- create pull request

martinbradford, 2017-08-10:
Hmmm, not too happy about tinkering with your repository! You sure you want me to do this? The files in question are "alexa.c" and "alexa_speech_recogniser.c" - in each case, simply update "freertos.h" to be "FreeRTOS.h"

ESP32DE, 2017-08-10:
@martinbradford 
``alexa_speech_recognizer.c``
you mean this [code line](https://github.com/MrBuddyCasino/ESP32_Alexa/blob/master/components/alexa/alexa_speech_recognizer.c#L13)?

``#include "freertos/freertos.h"`` updating to ``#include "freertos/FreeRTOS.h"``

----------

``alexa.c``
if you mean this [code line](https://github.com/MrBuddyCasino/ESP32_Alexa/blob/master/components/alexa/alexa.c#L18)

think there is no problem in this line?
``#include "freertos/FreeRTOS.h"``

----------

martinbradford, 2017-08-10:
I do - in the version I pulled down a few days ago, it needed to be updated in both files.

ESP32DE, 2017-08-10:
@martinbradford 

is this ok?

-----

martinbradford, 2017-08-10:
They both look right to me now. I'm just trying to build using an earlier version of BearSSL to see if that fixes the crash I've been getting - the 0.4 code seems to be quite dirty. I will pull a clean copy of your code later or tomorrow and check if it builds under Linux.

MrBuddyCasino, 2017-08-10:
Ok. I have to change the bearssl randomness injection anyway, the way I did it is not how it is supposed to be done.

martinbradford, 2017-08-10:
Ah, ok... my experiments with building with earlier releases of BearSSL have introduced more problems than they have solved!

mikey60, 2017-08-11:
I also get the same errors when I run it and the same GPIO[0] intr, val: 1 Guru Meditation Error when I press the GPIO0 button.

MrBuddyCasino, 2017-08-11:
You're not the only one having trouble it seems. Must be frustrating, I'm sorry. Hopefully I'll find some time today to have a look.

martinbradford, 2017-08-11:
no need to apologise! This is your experimental code that you are letting the rest of us use free of charge - no support responsibility!

I'm just sorry that I don't know enough about the ESP32 platform to be much use at debugging just yet - but I am learning fast...

MrBuddyCasino, 2017-08-11:
I think I fixed the bug. It was indeed an error with BearSSL entropy injection and a change in the git submodule that went unnoticed by me. Apologies for wasting everyones time.

martinbradford, 2017-08-11:
As I said before - no need to apologise - that goes with charging for the code! Will test shortly and report back

martinbradford, 2017-08-11:
Now working fine! However, "freertos.h" in auth_handler.c seems to have gone back to lower case...

MrBuddyCasino, 2017-08-11:
Weird. Fixed it. To confirm, voice commands are working now?

martinbradford, 2017-08-11:
It's all working fine - I can hit the button, ask "What's the time" and it replies "The time is 11am". I get a bit of crackling in the speaker while it's recognising speech - I guess it would be possible to mute the output channel until it has pinged the sample off to the cloud... The speech is also a bit "micky mouse" - pitch seems high and it is a bit fast - I'm guessing that there is some discrepancy in the sample rates across the piece - but that sort of thing can be resolved....

MrBuddyCasino, 2017-08-11:
Oh that was actually intentional, I sped the voice up for my Alexa rabbit. I should make it configurable via menuconfig.
You can change it in app_main.c like this:
`renderer_config->sample_rate_modifier = 1.0;`

Same thing for local echo, disable it alexa_speech_recognizer.c line 118, I should add a menuconfig option.

martinbradford, 2017-08-11:
And now she is back to her seductive self! Excellent work!

martinbradford, 2017-08-11:
P.S. is it up to me to close this issue? It looks like it is all fixed now...

Next step is to implement the "trigger word" - I saw a short video from Espressive which demonstrated that they know how to do it, but I have not seen any code from them!

MrBuddyCasino, 2017-08-11:
It might be possible with the Wrover module with the additional 4MB pSRAM and the Snowboy project (see https://github.com/Kitt-AI/snowboy), but IMHO not possible with the Wroom module. That would be a welcome addition indeed.

I'm closing this issue, feel free to open another one for wakeword detection.

closed 2017-08-11 by MrBuddyCasino

mikey60, 2017-08-11:
Yes, there is no need to apologise. I just tried your latest version and it
works.  Thanks for the all the hard work you put into the project.

martinbradford, 2017-08-11:
I am currently trying to track down the code for the locally hosted wake up word recognition. There is a demo of it working in Chinese on YouTube and a suggestion that it will be available in English some time soon. I'll report back if I get a reply from Espressive

MrBuddyCasino, 2017-08-11:
Cool! Just open a separate issue if you find it, post it at: https://www.esp32.com/viewtopic.php?f=17&t=2618

There are Espressif staff there that may be able to help.

martinbradford, 2017-08-14:
I got a reply from Espressif - they were not happy with the performance of their initial code and are working on a rewrite - hope to have something within the next few weeks!

hchaudhary1, 2017-08-14:
@martinbradford did they release any schematics for this?