MrBuddyCasino/ESP32_Alexa #6: fatal error: brssl.h: No such file or directory

teamroster, 2017-08-19:
When I do a make, the following error occurs. I'm able to compile and run your MP3 decoder project.

...
CC alexa_speech_recognizer.o
CC auth_handler.o
CC event_send_settings_updated.o
CC event_send_state.o
CC events_context.o
CC js_event_recognize.o
CC js_event_settings.o
CC js_event_sync_state.o
CC multipart_producer.o
CC stream_handler_directives.o
CC stream_handler_events.o
AR libalexa.a
CC app_trace.o
CC app_trace_util.o
AR libapp_trace.a
CC esp_ota_ops.o
AR libapp_update.a
CC asio.o
/Users/Steven/ESP32_Alexa/components/asio/./asio.c:22:19: fatal error: brssl.h: No such file or directory
compilation terminated.
make[1]: *** [asio.o] Error 1
make: *** [component-asio-build] Error 2

If I just copy brssl.h so it will be found then the following error:
    brssl.h:33:21: fatal error: bearssl.h: No such file or directory
This file is not in any of the bear_ssl directories.

Any suggestions? 
thanks

hchaudhary1, 2017-08-19:
did you init the submodule?

MrBuddyCasino, 2017-08-19:
yup, you need to do:

git submodule init && git submodule update