MrBuddyCasino/ESP32_Alexa #7: Deep sleep, wake up on button press function?

mano1979, 2017-10-01:
Is it possible to implement deep sleep when not active? And only after button press it wakes up and listens to commands? 

This would be great for battery powered esp32 alexa clients.

MrBuddyCasino, 2017-10-03:
You need a persistent connection to the Alexa backend before you can do anything, and I don't think you can maintain a wifi connection in deep sleep. So I'd say this is not possible.

mano1979, 2017-10-04:
@MrBuddyCasino I think the low power 2nd cpu can keep the connection going. Though, even if not, can't the esp32 simply reconnect to the wifi and connect to the alexa server, when woken up?

MrBuddyCasino, 2017-10-04:
Do you mean the 2nd core or the ULP coprocessor? The ULP can only do very simple tasks.
Reconnecting takes a few seconds due to the SSL handshake and probably the wifi connection itself, so its not instantaneous. If you can't maintain the connection, you might as well just cut the power.

You could try deep sleep and caching the SSL session tickets for a faster reconnect, but that only makes sense if the wifi connection stays up, which would be pointless due to the energy consumption AFAIK.

mano1979, 2017-10-04:
Why would the wifi have to stay up? Reconnecting doesn't have to take long. Depending on the settings. Even though, a 2 second wait after pressing the button before i speak is fine by me.

I found this post that shows some settings to speed things up after a wake. https://esp32.com/viewtopic.php?f=2&t=1238

MrBuddyCasino, 2017-10-04:
Thats faster than I thought. Feel free to implement something, if its decent code I usually accept pull requests.

mano1979, 2017-10-04:
I have no idea where to start. I am not a great programmer. More an engineer. 

I will ask my friend to help me though. I will do a pull request when i have something interesting.

Thanks for your quick reply!

MrBuddyCasino, 2017-10-04:
You're welcome.

closed 2017-10-04 by MrBuddyCasino