MrBuddyCasino/ESP32_Alexa #9: Touch instead of button & led function.

mano1979, 2017-11-02:
Can you tell.me what i need to change to use a touch sensor (build-in) instead of the button on gpio0?

Also i want to adjust the led function so i can power-off the i2s amplifier module while not in use. This will.save a lot of power. Where and what would i need to change for that? (I just want to drive a mosfet/transistor to make or break the power to the amplifier, at the moment the voice feedback is finished, and back on when there should be sound.

MrBuddyCasino, 2017-11-02:
I didn't use the CapSense peripherals yet, so I can't help you there.

If it is just about enabling/disabling of the I2S amp, some have a special purpose pin for that. You could maybe try to hook into the audio_renderer module for start/end signals.

mano1979, 2017-11-02:
I have ordered the i2s amp from adafruit. Along with their microphone module. I believe it has an enable pin but am not sure if it cuts the power completely. Also i am not sure if , after a power-down, you have to reconnect/re-initialize the i2s connection.

I found this code in the audio renderer module and think this is the place to enable/disable the amp. But can you explain what function does exactly what? Or in other words: when does the audio start, stop and is removed(?)?

void renderer_start()
{
    if(renderer_status == RUNNING)
        return;

    renderer_status = RUNNING;
    i2s_start(renderer_instance->i2s_num);

    // buffer might contain noise
    i2s_zero_dma_buffer(renderer_instance->i2s_num);
}

void renderer_stop()
{
    if(renderer_status == STOPPED)
        return;

    renderer_status = STOPPED;
    i2s_stop(renderer_instance->i2s_num);
}

void renderer_destroy()
{
    renderer_status = UNINITIALIZED;
    i2s_driver_uninstall(renderer_instance->i2s_num);
}

MrBuddyCasino, 2017-11-02:
Actually I'm not 100% sure the lifecycle is what you want, meaning that the renderer is only active during voice in/out, because its some time since I last worked on that.

Maybe add a printf() to those functions and see when they are called. If its alive for too long for your purposes, there will be changes required in the app to shut down the renderer when unused. If so, you'll have to figure those out for yourself I'm afraid.

mano1979, 2017-11-02:
Ok. I'll have a go at it once my hardware has arrived.

By the way, if you are interested in the touch sensors, i found this: http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/touch_pad.html

It explains it all.

mano1979, 2017-11-03:
Assuming that one of the leds goes on while audio is beeing played and goes off after it is finished (like with alexapi), maybe that would be a better place to change the code for my function?

Can you tell me in what file this is done?

MrBuddyCasino, 2017-11-03:
That would be led_ui.c.

closed 2018-01-02 by MrBuddyCasino