MrBuddyCasino/ESP32_Alexa #10: Upload 10ms 320 bytes per chunk to AVS

hchaudhary1, 2017-12-10:
Per the AVS documentation, it is recommend to upload 10ms aka 320 bytes per chunk while upload audio.

reviewing the code, it seems that this is not observed:
https://github.com/MrBuddyCasino/ESP32_Alexa/blob/master/components/alexa/alexa_speech_recognizer.c#L115

MrBuddyCasino, 2017-12-10:
Yes I know, this is to decrease latency. But using 320 bytes creates an awful lot of overhead per packet and leads to throughput issues on the ESP32, so by trial and error I arrived at the current values.

hchaudhary1, 2017-12-10:
got it.
it looks like, you are transmitting as many samples as possible. I suppose it makes sense to let AVS buffer the data rather than an extremely constrained device like an ESP32

closed 2017-12-10 by hchaudhary1