MrBuddyCasino/ESP32_Alexa #11: Nothing coming back from AVS?

mano1979, 2017-12-30:
When i press the button, the neopixels don't come on, and when i ask something, i get no reaction.
I am using the ics-43434 mic and internal dac output. Below is the serial output:

`
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5544
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:13716
entry 0x40078fb4
[0;32mI (30) boot: ESP-IDF v3.1-dev-137-g4f44d16 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 01:04:33[0m
[0;32mI (30) boot: Enabling RNG early entropy source...[0m
[0;32mI (35) boot: SPI Speed      : 40MHz[0m
[0;32mI (40) boot: SPI Mode       : DIO[0m
[0;32mI (44) boot: SPI Flash Size : 4MB[0m
[0;32mI (48) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (81) boot: End of partition table[0m
[0;32mI (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x72628 (468520) map[0m
[0;32mI (257) esp_image: segment 1: paddr=0x00082650 vaddr=0x3ffb0000 size=0x06244 ( 25156) load[0m
[0;32mI (267) esp_image: segment 2: paddr=0x0008889c vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (268) esp_image: segment 3: paddr=0x00088ca4 vaddr=0x40080400 size=0x0736c ( 29548) load[0m
[0;32mI (287) esp_image: segment 4: paddr=0x00090018 vaddr=0x400d0018 size=0xedb3c (973628) map[0m
[0;32mI (625) esp_image: segment 5: paddr=0x0017db5c vaddr=0x4008776c size=0x0b2fc ( 45820) load[0m
[0;32mI (644) esp_image: segment 6: paddr=0x00188e60 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (656) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (656) boot: Disabling RNG early entropy source...[0m
[0;32mI (657) cpu_start: Pro cpu up.[0m
[0;32mI (660) cpu_start: Starting app cpu, entry point is 0x40081060[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (671) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (678) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (684) heap_init: At 3FFC1BD0 len 0001E430 (121 KiB): DRAM[0m
[0;32mI (690) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (696) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (703) heap_init: At 40092A68 len 0000D598 (53 KiB): IRAM[0m
[0;32mI (709) cpu_start: Pro cpu start user code[0m
[0;32mI (56) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (57) main: starting app_main()[0m
[0;33mW (57) main: 168: - RAM left 239196[0m
[0;31mE (57) main: MAC address: 30:ae:a4:37:f1:74[0m
[0;32mI (97) main: hardware initialized[0m
[0;32mI (97) renderer: init I2S mode 2, port 0, 16 bit, 44100 Hz[0m
[0;32mI (97) I2S: queue free spaces: 1[0m
[0;32mI (97) I2S: DMA Malloc info, datalen=blocksize=256, dma_buf_count=32[0m
[0;32mI (107) I2S: PLL_D2: Req RATE: 44100, real rate: 2777.000, BITS: 16, CLKM: 30, BCK: 60, MCLK: 30.234, SCLK: 88864.000000, diva: 64, divb: 14[0m
[0;32mI (117) I2S: DMA Malloc info, datalen=blocksize=512, dma_buf_count=32[0m
[0;32mI (127) I2S: PLL_D2: Req RATE: 16000, real rate: 16025.000, BITS: 32, CLKM: 39, BCK: 4, MCLK: 4096000.000, SCLK: 1025600.000000, diva: 64, divb: 4[0m
[0;33mW (137) main: 194: - RAM left 193168[0m
[0;32mI (137) alexa: 572: - RAM left 193168[0m
[0;32mI (147) asio: adding task: led_ui[0m
I (157) wifi: wifi firmware version: 275023b
I (157) wifi: config NVS flash: enabled
I (157) wifi: config nano formating: disabled
[0;32mI (167) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (177) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (197) wifi: Init dynamic tx buffer num: 32
I (197) wifi: Init data frame dynamic rx buffer num: 32
I (197) wifi: Init management frame dynamic rx buffer num: 32
I (197) wifi: wifi driver task: 3ffd2898, prio:23, stack:4096
I (197) wifi: Init static rx buffer num: 10
[... 785 lines cut ...]

[0;32mI (39057) alexa: All headers received for stream 7[0m
events data:
--------abcde123

Content-Type: application/json; charset=UTF-8

{"directive":{"header":{"namespace":"System","name":"SetEndpoint","messageId":"cf7568c5-df5e-482d-b210-1e67c476266c","dialogRequestId":"1"},"payload":{"endpoint":"https://avs-alexa-na.amazon.com"}}}

--------abcde123--

on_part_data_begin
on_header_field Content-Type
on_header_value application/json; charset=UTF-8
directive detected
on_headers_complete
on_part_data_end
handle_directive:
{"directive":{"header":{"namespace":"System","name":"SetEndpoint","messageId":"cf7568c5-df5e-482d-b210-1e67c476266c","dialogRequestId":"1"},"payload":{"endpoint":"https://avs-alexa-na.amazon.com"}}}
on_body_end
[0;32mI (39177) alexa: frame received: 0[0m
[0;32mI (39207) alexa: frame received: 0[0m
[0;32mI (39207) alexa: closed stream 7 with error_code=0[0m
[0;32mI (39207) asio_http2: closed stream 7 with error_code=0[0m
`

MrBuddyCasino, 2017-12-30:
You can listen to the audio in the Alexa app. Does it sound right?

mano1979, 2017-12-30:
The app says it is not online. However, i saw that same message with AlexaPi on a raspberry pi, wich worked great.

Is the app supposed to see the alexa client by itself?

mano1979, 2017-12-30:
I managed to fix the "Offline" issue by changing the server location from europe to US. Nkw it shows online. 

But that is it. I see no commands in the app (voice recordings)

mano1979, 2018-01-01:
Here is the output from booting:
```ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5544
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:13716
entry 0x40078fb4
[0;32mI (30) boot: ESP-IDF v3.1-dev-137-g4f44d16 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 06:11:43[0m
[0;32mI (30) boot: Enabling RNG early entropy source...[0m
[0;32mI (36) boot: SPI Speed      : 40MHz[0m
[0;32mI (40) boot: SPI Mode       : DIO[0m
[0;32mI (44) boot: SPI Flash Size : 4MB[0m
[0;32mI (48) boot: Partition Table:[0m
[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (74) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (82) boot: End of partition table[0m
[0;32mI (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x72638 (468536) map[0m
[0;32mI (257) esp_image: segment 1: paddr=0x00082660 vaddr=0x3ffb0000 size=0x06244 ( 25156) load[0m
[0;32mI (267) esp_image: segment 2: paddr=0x000888ac vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (268) esp_image: segment 3: paddr=0x00088cb4 vaddr=0x40080400 size=0x0735c ( 29532) load[0m
[0;32mI (287) esp_image: segment 4: paddr=0x00090018 vaddr=0x400d0018 size=0xedb3c (973628) map[0m
[0;32mI (626) esp_image: segment 5: paddr=0x0017db5c vaddr=0x4008775c size=0x0b30c ( 45836) load[0m
[0;32mI (645) esp_image: segment 6: paddr=0x00188e70 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (657) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (657) boot: Disabling RNG early entropy source...[0m
[0;32mI (658) cpu_start: Pro cpu up.[0m
[0;32mI (661) cpu_start: Starting app cpu, entry point is 0x40081060[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (672) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (678) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (684) heap_init: At 3FFC1BD0 len 0001E430 (121 KiB): DRAM[0m
[0;32mI (691) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (697) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (703) heap_init: At 40092A68 len 0000D598 (53 KiB): IRAM[0m
[0;32mI (710) cpu_start: Pro cpu start user code[0m
[0;32mI (56) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (58) main: starting app_main()[0m
[0;33mW (58) main: 168: - RAM left 239196[0m
[0;31mE (58) main: MAC address: 30:ae:a4:38:81:2c[0m
[0;32mI (98) main: hardware initialized[0m
[0;32mI (98) renderer: init I2S mode 2, port 0, 16 bit, 44100 Hz[0m
[0;32mI (98) I2S: queue free spaces: 1[0m
[0;32mI (98) I2S: DMA Malloc info, datalen=blocksize=256, dma_buf_count=32[0m
[0;32mI (108) I2S: PLL_D2: Req RATE: 44100, real rate: 2777.000, BITS: 16, CLKM: 30, BCK: 60, MCLK: 30.234, SCLK: 88864.000000, diva: 64, divb: 14[0m
[0;32mI (118) I2S: DMA Malloc info, datalen=blocksize=512, dma_buf_count=32[0m
[0;32mI (128) I2S: PLL_D2: Req RATE: 16000, real rate: 16025.000, BITS: 32, CLKM: 39, BCK: 4, MCLK: 4096000.000, SCLK: 1025600.000000, diva: 64, divb: 4[0m
[0;33mW (138) main: 194: - RAM left 193168[0m
[0;32mI (138) alexa: 572: - RAM left 193168[0m
[0;32mI (148) asio: adding task: led_ui[0m
I (158) wifi: wifi firmware version: 275023b
I (158) wifi: config NVS flash: enabled
I (158) wifi: config nano formating: disabled
[0;32mI (168) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (178) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (188) wifi: Init dynamic tx buffer num: 32
[... 157 lines cut ...]

--nghttp2123456789--

[0;32mI (8878) nghttp2: Response headers for stream ID=5:[0m
:status: 204
access-control-allow-origin: *
x-amzn-requestid: 06c4defffe645d2e-00002200-0007b7b0-ec599f86d651de3a-c768d897-5

[0;32mI (8908) alexa: All headers received for stream 5[0m
[0;32mI (8908) alexa: closed stream 5 with error_code=0[0m
[0;32mI (8918) asio_http2: closed stream 5 with error_code=0[0m
[0;32mI (9388) mad_decoder: decoder stopped[0m
[0;32mI (9388) mad_decoder: MAD decoder stack: 724
[0m
[0;33mW (9398) led_ui: switching anim[0m
[0;31mE (9398) led_ui: setColors 0=-2146288040, 1=1073545920[0m
[0;32mI (9778) nghttp2: Response headers for stream ID=3:[0m
:status: 204
access-control-allow-origin: *
x-amzn-requestid: 06c4defffe645d2e-00002200-0007b7b0-ec599f86d651de3a-c768d897-3

[0;32mI (9808) alexa: All headers received for stream 3[0m
[0;32mI (9808) alexa: closed stream 3 with error_code=0[0m
[0;32mI (9818) asio_http2: closed stream 3 with error_code=0[0m
```

MrBuddyCasino, 2018-01-02:
Not sure about the "closed stream n with error_code=0", but the rest looks alright. What happens if you press GPIO0 and ask something?

mano1979, 2018-01-02:
When i press the button: Leds blink red and blue, and some noise is coming from the speaker. It seems to send something (see output first post) but nothing comes back. No answer, nothing.

mano1979, 2018-01-02:
Have you also seen the "204" response code?
204 means "no data". (Http error codes)

So that would mean there is nothing coming back from amazon?

mano1979, 2018-01-03:
I found the issue!! It works now!

On my board, gpio17 is not numbered. While searchin online, i found a refenrence that now, seems to be wrong.
It said that gpio17 was named tx0. Now it seems that it accually should be tx2. So i changed the data pin to tx2 and it works!

Only issue is the voice is very fast. Chipmunk style!
Can i change that somewhere?

mano1979, 2018-01-03:
I found the sample rate multiplier for the output sound. That is now set to 1 instead of 1.25. 

But the recorded voice, my voice is still in chipmunk. (Listened back in the alexa app). Where can i change the sample rate of the mic input?

MrBuddyCasino, 2018-01-04:
Hmm, your recorded voice shouldn't be high-pitched. AFAIK I'm using the two different I2S peripherals for that reason.

mano1979, 2018-01-04:
After listening again, it is not high pitched, just speed up. 

What do you mean with "two different i2s peripherals"? Are you using 2 microphones?

MrBuddyCasino, 2018-01-04:
No, but I2S is full-duplex interface. You can do playback and recording on the same port, but then the bitclock must be the same. Thats why I use the separate I2S_0 and I2S_1 periphs on the ESP32.

mano1979, 2018-01-04:
Aha! So if i understand correctly, i might be using the wrong i2s port?

Or is it just the bitclock that is wrong? Can this be fixed by switching around the pins i connect the microphone to?

MrBuddyCasino, 2018-01-04:
No, I was just explaining that the input & output are independent, this shouldn't happen. But I don't have the hardware with me right now to make tests.

mano1979, 2018-01-04:
Ok. No problem. Let me know when you know something. 

My thoughts where that it might have something to do with the difference between the ics43434 and the sph0645 microphone. Since they seem to have slightly different outputs (24bit vs 32bit). But like i said, i am not a great programmer, and bit's and byte's are waaaay out of my league.

MrBuddyCasino, 2018-01-04:
I tested both ics43434 and sph0645, both were working fine for me (the ics43434 sounds better).

mano1979, 2018-01-04:
Yes. I use the ics43434 too. 

Weirdly enough though, the commands i give are recognized most of the time. Even though the audio is about twice the speed.

Thank you for your help!

junTMUG, 2018-01-15:
Modifying a line in send_speech_read_callback function in alexa_speech_recognizer.c will make better recoded sound quality and better recognition result for me.

In case AUDIO_DATA:
I changed the line from "if(rounds > 1) {" to "if(rounds > 4) {"

mano1979, 2018-01-15:
Tried it, but this doesn't change the speed of the microphone audio. In the Alexa app, my voice still sounds sped up.

Don't know if the audio got better. I never had any issues with it recognizing my commands. (Except for the occasional time i slurry my words. English is not my native language)

closed 2018-01-30 by MrBuddyCasino