MrBuddyCasino/ESP32_Alexa #13: Random resets panic crash

mano1979, 2018-01-04:
about every 10 to 15 minutes the code crashes and the esp32 is rebooted. Te histerical laugh almost giving me a heartattack. Crash report is below:
```[0;32mI (669308) alexa: frame received: 7[0m
Guru Meditation Error: Core  1 panic'ed (LoadStoreError)
. Exception was unhandled.
Register dump:
PC      : 0x401b4443  PS      : 0x00060c30  A0      : 0x80128af9  A1      : 0x3ffcff70  
A2      : 0x40000000  A3      : 0x3ffd0170  A4      : 0x000000a2  A5      : 0x3aa50aa2  
A6      : 0x0a050a02  A7      : 0x32210210  A8      : 0x56fbc95f  A9      : 0x0000001f  
A10     : 0x6aa56f12  A11     : 0xf0f0f0f0  A12     : 0xf3ed8a9d  A13     : 0x0a050f02  
A14     : 0x030d0a0d  A15     : 0x060b090f  SAR     : 0x00000010  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x401b4443:0x3ffcff70 0x40128af6:0x3ffcff90 0x401280e9:0x3ffd01e0 0x40128336:0x3ffd0210 0x401b42d1:0x3ffd0250 0x40125f83:0x3ffd0280 0x401263b9:0x3ffd02a0 0x40123e25:0x3ffd02c0 0x40123e55:0x3ffd0310 0x40122733:0x3ffd0330 0x401227a1:0x3ffd0350 0x40121892:0x3ffd0370 0x4010ca13:0x3ffd03a0

Rebooting...
ets Jun  8 2016 00:22:57

```

MrBuddyCasino, 2018-01-04:
I'm afraid the ping to prevent timeouts isn't implemented yet, so thats what happens. AWS closes the connection and the whole thing crashes.

mano1979, 2018-01-04:
Ah i see. Will this be implemented anytime soon?

MrBuddyCasino, 2018-01-04:
I've been meaning to do that since months, so probably not. :-)
I'm quite busy learning other tech stuff atm due to professional obligations. Something like an asio timer task would have to be implemented which sends the http2 ping packet, driven by the ESP32 hardware timers.

mano1979, 2018-01-04:
Ok. Please keep this issue open for now. I will ask a friend to have a look at it. He is much more a programmer than me. I will post back any findings or questions i have here.

flakeydabanana, 2018-01-10:
It looks like the ping is already implemented, but the timer never gets to 300s
`		if(difftime(time(NULL), last_action) > 4 * 60) {
		nghttp2_submit_ping(h2, NGHTTP2_FLAG_NONE, NULL);
		time(&last_action);
		ESP_LOGI(TAG, "Sending keepalive ping....");
		}`

I guess a better way would be to handle the disconnection and then just reconnect

mano1979, 2018-01-10:
#flakeydabanana Would you know how to do that? I am not much of a programmer myself.

MrBuddyCasino, 2018-01-11:
@flakeydabanana I suppose you got that from components/nghttp_client/nghttp2_client.c? This is not used anymore, I should clean it up. I changed the programming model from threaded to async, as I needed to reclaim the stack memory. See components/asio/asio_http2.c.

mano1979, 2018-01-11:
Is that whole file used for the "ping" function or is it a specific block? I couldn't find anything recognizable in there.

chegewara, 2018-06-01:
I have implemented ping functionality fairly easy. Just i have to do some more test to see why code is crashing after about 50 minutes. @MrBuddyCasino If you wish i can post code or prepare PR.

MrBuddyCasino, 2018-06-01:
Nice! I suppose a PR is easiest.

chegewara, 2018-06-01:
Based on your code i started this task in alexa.c:

```
#define TICKS_TO_DELAY 4 * 60 * 1000
void delayed_server_ping_task(void* p){
    TickType_t last_wake_time;
    TickType_t ticks_before_delay;

    last_wake_time = xTaskGetTickCount();
    ticks_before_delay = last_wake_time;

    while(1){
        vTaskDelayUntil(&last_wake_time, TICKS_TO_DELAY );
        event_send_ping(alexa_session);
    }
}
```
to send GET request to this address:
`static char *uri_ping = ALEXA_ENDPOINT "/ping";`

There is some more code involved, but i think you know what i mean.

mano1979, 2018-06-02:
@chegewara great work!! i can't wait to test it out. When can we expect your pull request?

mano1979, 2018-06-04:
@chegewara I downloaded your fork of ESP32-Alexa and now get resets every 10 minutes or so, sometimes a bit longer.
I thought that these resets might happen because of a slight offset in the timer of the esp. So i wanted to adjust the time between pings to 4m40s or so. But while i was searching through the code, i wasn't able to find the above code in alexa.c.
Did you embed this code in your master branch yet? and if not, how can it be that my connection stays alive longer than before with @MrBuddyCasino code?

chegewara, 2018-06-04:
Hi @mano1979, yes my ping request code to keep connection open is pushed to github. I have one more addition in my code that is reconnecting when amazon server sends GOAWAY frame. I will try to clean my code and push it to github.

Im guessing now, but maybe your code is crashing due to difference in sdkconfig. If you could provide info about decoded crash info then we can try to find a reason. I am using this to decode backtrace:
https://github.com/me-no-dev/EspExceptionDecoder

Its quick and easy to use.

mano1979, 2018-06-04:
@chegewara I tried to install the exception decoder but had trouble using it.

however, this is what i got from the serial terminal on crash (and part of boot sequence)

`[0;32mI (669134) alexa: frame received: 7[0m
Guru Meditation Error: Core  1 panic'ed (LoadStoreError)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x401c1267  PS      : 0x00060230  A0      : 0x800ede35  A1      : 0x3ffd11e0  
A2      : 0x40000000  A3      : 0x3ffd13e0  A4      : 0x00000086  A5      : 0x46fed386  
A6      : 0x060e0306  A7      : 0x13310001  A8      : 0xe83a43a3  A9      : 0x0000001f  
A10     : 0x76784615  A11     : 0xf0f0f0f0  A12     : 0x177fcd08  A13     : 0x06080605  
A14     : 0x070f0d08  A15     : 0x080a0303  SAR     : 0x00000010  EXCCAUSE: 0x00000003  
EXCVADDR: 0x40000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x401c1267:0x3ffd11e0 0x400ede32:0x3ffd1200 0x400ed425:0x3ffd1450 0x400ed672:0x3ffd1480 0x401c10f5:0x3ffd14c0 0x400eb2bf:0x3ffd14f0 0x400eb6f5:0x3ffd1510 0x400e9249:0x3ffd1530 0x400e9279:0x3ffd1580 0x400e7d17:0x3ffd15a0 0x400e7d85:0x3ffd15c0 0x400e6ef5:0x3ffd15e0 0x400d3fdf:0x3ffd1610

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5716
load:0x40078000,len:0
load:0x40078000,len:14960
entry 0x4007862c
[0;32mI (28) boot: ESP-IDF v3.1-dev-1193-g64b56be 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 23:08:16[0m
[0;32mI (29) boot: Enabling RNG early entropy source...[0m
[0;32mI (35) boot: SPI Speed      : 40MHz[0m
[0;32mI (39) boot: SPI Mode       : DIO[0m
[0;32mI (43) boot: SPI Flash Size : 4MB[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (80) boot: End of partition table[0m
`

chegewara, 2018-06-04:
There is not much info here. Thats why im using this tool (its arduino tool, but can be used with esp-idf). When you start this tool it will ask for elf file. That file is ESP32_alexa/build/elaxa.elf, then copy/paste this:
`Backtrace: 0x401c1267:0x3ffd11e0 0x400ede32:0x3ffd1200 0x400ed425:0x3ffd1450 0x400ed672:0x3ffd1480 0x401c10f5:0x3ffd14c0 0x400eb2bf:0x3ffd14f0 0x400eb6f5:0x3ffd1510 0x400e9249:0x3ffd1530 0x400e9279:0x3ffd1580 0x400e7d17:0x3ffd15a0 0x400e7d85:0x3ffd15c0 0x400e6ef5:0x3ffd15e0 0x400d3fdf:0x3ffd1610
`

mano1979, 2018-06-04:
yep, got it

```Decoding stack results
0x401c1267: xorbuf at /home/mano/esp/ESP32_Alexa_Ping/components/bear_ssl/src/symcipher/aes_ct_ctr.c line 45
0x400ede32: br_aes_ct_ctr_run at /home/mano/esp/ESP32_Alexa_Ping/components/bear_ssl/src/symcipher/aes_ct_ctr.c line 98
0x400ed425: do_ctr at /home/mano/esp/ESP32_Alexa_Ping/components/bear_ssl/src/ssl/ssl_rec_gcm.c line 118
0x400ed672: gcm_encrypt at /home/mano/esp/ESP32_Alexa_Ping/components/bear_ssl/src/ssl/ssl_rec_gcm.c line 206
0x401c10f5: sendpld_flush at /home/mano/esp/ESP32_Alexa_Ping/components/bear_ssl/src/ssl/ssl_engine.c line 864
0x400eb2bf: sendpld_ack at /home/mano/esp/ESP32_Alexa_Ping/components/bear_ssl/src/ssl/ssl_engine.c line 889
0x400eb6f5: br_ssl_engine_sendapp_ack at /home/mano/esp/ESP32_Alexa_Ping/components/bear_ssl/src/ssl/ssl_engine.c line 1146
0x400e9249: asio_ssl_run_engine at /home/mano/esp/ESP32_Alexa_Ping/components/asio/asio_secure_socket.c line 1079
0x400e9279: asio_io_handler_ssl at /home/mano/esp/ESP32_Alexa_Ping/components/asio/asio_secure_socket.c line 1096
0x400e7d17: asio_registry_poll_connection at /home/mano/esp/ESP32_Alexa_Ping/components/asio/asio.c line 105
0x400e7d85: asio_registry_poll at /home/mano/esp/ESP32_Alexa_Ping/components/asio/asio.c line 152
0x400e6ef5: alexa_init at /home/mano/esp/ESP32_Alexa_Ping/components/alexa/alexa.c line 660
0x400d3fdf: alexa_task at /home/mano/esp/ESP32_Alexa_Ping/main/app_main.c line 50```

chegewara, 2018-06-04:
This is question to @MrBuddyCasino , i dont know this stack enough to say what its the reason. Ive got it too, but in my case its after about 50 minutes (now maybe even later).

mano1979, 2018-06-05:
@chegewara Do you also have this issue that your voice is recorded in high speed?
When i play back my commands through the alexa android app, the audio is sped up. So yesterday i tried to set the cpu frequency to 240mhz instead of the default 160mhz. Allthough the entire system works faster, the audio (commands) got faster too. now that i'm back on 160mhz, the commands are so screwed up that they are not recognized anymore. (while it used to work before, even though sped up)

Do you have dual microphones or a single one? If dual, how did you wire them?

MrBuddyCasino, 2018-06-05:
@mano1979 I suspect this crash is due to some heap corruption or out of memory condition

chegewara, 2018-06-05:
I am using ESP32 simple alexa board (newest version i think) and i dont have any issue with recording my voice, neither with 240MHz or 160 MHz. Its one mic board. To make alex speak with right pitch i have to change this value to about 0.54:
https://github.com/MrBuddyCasino/ESP32_Alexa/blob/master/main/app_main.c#L98

Its due to conversion from 44100 to 24000 (24000/44100).

chegewara, 2018-06-05:
This is my latest log after about 5 hours or so (no longer crashing). Now i have to handle GOAWAY frame:
```
E (18600416) nghttp2: Could not submit HTTP request: Protocol error
I (18600426) nghttp2: failed to submit request
E (18900346) alexa: PING
I (18900346) nghttp2: new nghttp stream, uri: https://avs-alexa-na.amazon.com/ping
I (18900346) nghttp2: Request headers:
I (18900346) nghttp2: :method: GET
I (18900346) nghttp2: :scheme: https
I (18900356) nghttp2: :authority: avs-alexa-na.amazon.com
I (18900356) nghttp2: :path: /ping
I (18900366) nghttp2: authorization: Bearer Atza|IwEBIHArSA6Z51oiTEIF6u7RQi0FdLx0OiK_IZjo9QN7dqdUEbU5n7d5u8lmjpTHnQ-ujR8uGov61sY8NOcZr3aEiyr3ZBKhfWWl_V83nPpNUYCKqL2RjvYwwiFq58WkCq70xYPZw4MFrqIqceTSJdzY_OjszZRm9mFZC8SIsJ8D4KPlI5GwYHFwbvbM59IJi3pKbO5XK7Mbc8bCNPsiRsiKix7KXsfFHoBuKjannpXhzFi2ZP9_rZWvqMXF2ZLbiakhQ19PdyvZpeHLTtBlK4Wyd8pti-wu9lVbRcKn1Qoar3hNYUr3M20DWgQB-xMFkk8Uo7TN9_nn_uzEGJ5Qu_uGHOu2TOJWJrHkgajYAplGv0yIeb0iDUFUu66gjTsR42EHHwzIqXsl8thxNU0wO2a9nnueiurcWff9CdpNfEpSXb-Sq9PhTC2l45QQ6WVSbOtxavq8mndiv_Yb5E-zoI23C0hyQFo4VnsA6pqVf-Kryf7zg8YhhevOnQitW3HNq28jHzm5OS7MepVA9LRzJ4A27_RvTl3an7bK2V31cRm9mXervvTgy-_V8HhUp8s1wKQy144

E (18900416) nghttp2: Could not submit HTTP request: Protocol error
I (18900426) nghttp2: failed to submit request
E (19200346) alexa: PING
I (19200346) nghttp2: new nghttp stream, uri: https://avs-alexa-na.amazon.com/ping
I (19200346) nghttp2: Request headers:
I (19200346) nghttp2: :method: GET
I (19200346) nghttp2: :scheme: https
I (19200356) nghttp2: :authority: avs-alexa-na.amazon.com
I (19200356) nghttp2: :path: /ping
I (19200366) nghttp2: authorization: Bearer Atza|IwEBIHArSA6Z51oiTEIF6u7RQi0FdLx0OiK_IZjo9QN7dqdUEbU5n7d5u8lmjpTHnQ-ujR8uGov61sY8NOcZr3aEiyr3ZBKhfWWl_V83nPpNUYCKqL2RjvYwwiFq58WkCq70xYPZw4MFrqIqceTSJdzY_OjszZRm9mFZC8SIsJ8D4KPlI5GwYHFwbvbM59IJi3pKbO5XK7Mbc8bCNPsiRsiKix7KXsfFHoBuKjannpXhzFi2ZP9_rZWvqMXF2ZLbiakhQ19PdyvZpeHLTtBlK4Wyd8pti-wu9lVbRcKn1Qoar3hNYUr3M20DWgQB-xMFkk8Uo7TN9_nn_uzEGJ5Qu_uGHOu2TOJWJrHkgajYAplGv0yIeb0iDUFUu66gjTsR42EHHwzIqXsl8thxNU0wO2a9nnueiurcWff9CdpNfEpSXb-Sq9PhTC2l45QQ6WVSbOtxavq8mndiv_Yb5E-zoI23C0hyQFo4VnsA6pqVf-Kryf7zg8YhhevOnQitW3HNq28jHzm5OS7MepVA9LRzJ4A27_RvTl3an7bK2V31cRm9mXervvTgy-_V8HhUp8s1wKQy144

```

chegewara, 2018-06-06:
https://www.tindie.com/products/microwavemont/esp32-simple-alexa/

mano1979, 2018-06-06:
@chegewara That board is using the same microphone as me. so that is not the problem.

I don't get it......
When i first downloaded/cloned your repo and flashed it, even though the recording is very fast, it worked fine. But now it won't anymore. nothing is recognized and the recorded audio is even worse than before (listening in the alexa android app).

so i removed the entire repo from my harddrive and cloned it again to start with a fresh image. But no change.....   How can it work fine before and not now while i didn't change anything?

The code runs fine though, it is just the scrambled audio that is not recognized by the alexa servers.

mano1979, 2018-06-06:
Can my issues have something to do with me using the internal dac as audio output?

MrBuddyCasino, 2018-06-06:
That could be it. I think the I2S interface is supposed to behave in the same way, but maybe its not.

chegewara, 2018-06-06:
@mano1979 You can always try with this repo without changes ive made. Then you can confirm its issue with my code or not.

mano1979, 2018-06-06:
@MrBuddyCasino I also saw an option called "PDM" in the audio output menu. What does that do?

@chegewara Yes, i wil try that. But your code used to work fine before. So i doubt that the problem is in your code.

mano1979, 2018-06-06:
Wich pin is used for this output?

MrBuddyCasino, 2018-06-06:
Same pin I think.

mano1979, 2018-06-06:
ok. Flashing now......

mano1979, 2018-06-06:
ok. sound has a lot of noise but overall not bad. But i still am not recognized. only if i talk real slow, alexa understands what i'm saying

mano1979, 2018-06-06:
After trying the original code from "this" repo, i conclude that the issues i have with NOT beeing understood by alexa and the high speed audio from he microphone still are there.  Also with this repo i used to have no issues with beeing understood. Even with the high speed microphone rec. (allthough it has never been THIS fast).

Can my hardware setup , beeing on a breadboard , cause any capacitive issues?

chegewara, 2018-06-07:
I have clue why this crash might happen. 
https://github.com/MrBuddyCasino/ESP32_Alexa/issues/13#issuecomment-394506172
Authorization token needs to be refreshed every hour.

chegewara, 2018-06-07:
@mano1979 What is your log after alexa speaking?
`I2S: PLL_D2: Req RATE: 13056, real rate: 13297.000, BITS: 16, CLKM: 47, BCK: 8, MCLK: 3343127.653, SCLK: 425504.000000, diva: 64, divb: 55`

mano1979, 2018-06-07:
This is after me talking:
```\0x1b[0;32mI (34863) renderer: changing sample rate from 44100 to 16000\0x1b[0m
\0x1b[0;32mI (34863) I2S: PLL_D2: Req RATE: 20000, real rate: 1262.000, BITS: 16, CLKM: 66, BCK: 60, MCLK: 66.667, SCLK: 40384.000000, diva: 64, divb: 42\0x1b[0m
```

This is after Alexa talking:
```\0x1b[0;32mI (174963) audio_player: RAM left 91760\0x1b[0m
\0x1b[0;32mI (174973) mad_decoder: decoder start\0x1b[0m
\0x1b[0;31mE (174973) mad_decoder: dec err 0x0101 (lost synchronization)\0x1b[0m
\0x1b[0;32mI (174983) renderer: changing sample rate from 16000 to 24000\0x1b[0m
\0x1b[0;32mI (174983) I2S: PLL_D2: Req RATE: 30000, real rate: 1893.000, BITS: 16, CLKM: 44, BCK: 60, MCLK: 44.444, SCLK: 60576.000000, diva: 64, divb: 28\0x1b[0m

mano1979, 2018-06-07:
@chegewara  The above whas with the original code from Buddy. Below is the string with your fork:
```I2S: PLL_D2: Req RATE: 12480, real rate: 786.000, BITS: 16, CLKM: 106, BCK: 60, MCLK: 106.838, SCLK: 25152.000000, diva: 64, divb: 53\0x1b[0m

mano1979, 2018-06-07:
I set that value to 1.0 and now i am understood by alexa again. (though int he android app my voice still sounds like a chipmunk at warp-speed)

One question though. The playback speed in buddy's repo, is very fast (alexa's voice), but now i flashed your fork, he voice is still so fast, eventhough it was not before... i will check the code to see what hat is set to, just in case

chegewara, 2018-06-07:
In this repo original value is 1.25

mano1979, 2018-06-07:
I set the value for speech playback to 0.75 since it was 1.0 in your fork and 1.25 in Buddy's code.
But the voice playback is still fast like it is set to 1.25.

It somehow looks like the flash migh not be overwritten correctly?

mano1979, 2018-06-07:
changed 44100 to 24000 but  the speech is now even faster. now i cannot even make up what she said.

though.... It now sounds the same speed as my voice in the alexa android app
maybe some sort of relation there?

chegewara, 2018-06-08:
Maybe both parameters need to be adjusted. Ive got lucky and after few minutes(i had to find parameter and see which way it needs to be changed) my board was ready to go.

mano1979, 2018-06-08:
Ok. I will try to play with the values a bit.

Good to see you keeping it alive for 2 hours. Mine still resets after 12 minutes.

I'll keep you posted.

chegewara, 2018-06-12:
I dont know if i should open new issue for this because its related. Finaly i managed to run over 2 hours. Here is set big stack for created task and i found that 6 * 1024 is enough:
https://github.com/MrBuddyCasino/ESP32_Alexa/blob/master/main/app_main.c#L187
Even with 6kB stack still its about 1.5k free, now its time to go down to 5*1024 stack.
`W (7701949) alexa: alexa_handler stack: 1556` <-- free stack

chegewara, 2018-06-12:
There is one more part of code that is required to keep alexa alive (i cant make PR myself at the moment, sorry). Access token is valid only 3600 seconds and then amazon server is sending GO_AWAY frame. Again, its most likely not best possible solution, but it works for me:

```
static int on_frame_recv_callback(nghttp2_session *session,
        const nghttp2_frame *frame, void *user_data)
{
    http2_session_data_t *session_data = user_data;
    alexa_session_t *alexa_session = session_data->user_data;

    switch (frame->hd.type) {
        case NGHTTP2_HEADERS:
            print_headers(frame->headers.nva, frame->headers.nvlen);
            if (frame->headers.cat == NGHTTP2_HCAT_RESPONSE) {
                ESP_LOGI(TAG, "All headers received for stream %d",
                        frame->headers.hd.stream_id);

                if (frame->headers.hd.stream_id
                        == alexa_session->stream_directives->stream_id) {
                    // once all headers for the downchannel are received, we're clear
                    ESP_LOGI(TAG, "setting DOWNCHAN_CONNECTED_BIT");
                    xEventGroupSetBits(alexa_session->event_group,
                            DOWNCHAN_CONNECTED_BIT);
                }
            }
            break;

        case NGHTTP2_GOAWAY:
            ESP_LOGW(TAG, "%d: - RAM left %d", __LINE__, esp_get_free_heap_size());
            ESP_LOGI(TAG, "frame received: %u", frame->hd.type);
            nghttp2_submit_rst_stream(session, NGHTTP2_FLAG_NONE, alexa_session->stream_directives->stream_id, NGHTTP2_STREAM_CLOSED);
            alexa_session->stream_directives->status = CONN_CLOSED;
            xEventGroupClearBits(alexa_session->event_group,
                            DOWNCHAN_CONNECTED_BIT);
                                    auth_token_refresh(alexa_session);

            ESP_LOGW(TAG, "%d: - RAM left %d", __LINE__, esp_get_free_heap_size());
            asio_new_generic_task("downchannel", alexa_session->registry, on_auth_token_valid_cb, alexa_session->event_group, alexa_session);
            /* send initial state when downchannel is connected */
            asio_new_generic_task("send_initial_state", alexa_session->registry, on_downchan_connected_cb, alexa_session->event_group, alexa_session);
            ESP_LOGW(TAG, "%d: - RAM left %d", __LINE__, esp_get_free_heap_size());
            break;

        default:
            ESP_LOGI(TAG, "frame received: %u", frame->hd.type);
            break;
    }
    return 0;
}
```

MrBuddyCasino, 2018-06-12:
thanks, merged that too

chegewara, 2018-06-12:
The issue with GO_AWAY frame is that even if first works (after about one hour) next causes crash because lack of RAM. Ive found that some heap can be released as described here:
https://github.com/MrBuddyCasino/ESP32_Alexa/issues/13#issuecomment-396664377

As for now this seems to be enough to work for longer without crash.

And MrBuddy, thanks for this great piece of code. All i am doing is paid project thats why i am trying to share info but i cant make too much PR's.

MrBuddyCasino, 2018-06-12:
I understand, thx.

chegewara, 2018-06-13:
I dont want to open new issue just to show how it works (video on YT) and it is related to this issue(crash). Over 8 hours and still working:
```
E (29403294) alexa: PING
:status: 204
access-control-allow-origin: *
x-amzn-requestid: 021038fffed1de70-00002533-0008e088-71657edca9c72c00-f934d19a-23

E (29700295) alexa: PING
:status: 204
access-control-allow-origin: *
x-amzn-requestid: 021038fffed1de70-00002533-0008e088-71657edca9c72c00-f934d19a-25
```
The issue is memory leaking on access token refresh:
```
W (3037612) asio: 96: - RAM left 28808
........
W (23687034) asio: 96: - RAM left 17544
W (26752667) asio: 96: - RAM left 13992
```

chegewara, 2018-06-25:
New ping version looks and works way better than last one and which is most important does not cause memory leak. I will try to find time and prepare clean PR with it. Here is how it looks like:
``` 
nghttp2_session *session = alexa_session->stream_events->http2_session->h2_session;
nghttp2_submit_ping(session, NGHTTP2_FLAG_NONE, NULL);
```

chegewara, 2018-06-28:
Like i promissed there is refactored ping event. PR is ready to merge.