MrBuddyCasino/ESP32_Alexa #15: mdns error

andrzej-grala, 2018-01-28:
I'm getting mdns error during compilation:
mdns_task.c:28:5: error: unknown type name 'mdns_server_t'
I'm using latest (3.0) version sdk and version 5.2 toolchain.

MrBuddyCasino, 2018-01-29:
Thats due to recent changes in the SDK. You can just delete the mdns stuff, its not used anyway.

andrzej-grala, 2018-01-29:
Thanks. Now works.

For me works also:
`/*
 * mdns.c
 *
 *  Created on: 23.04.2017
 *      Author: michaelboeckling
 */

#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"

#include "mdns.h"
#include "wifi.h"

#define TAG "mDNS"
#define MDNS_HOSTNAME "esp32-radio"
#define MDNS_INSTANCE "ESP32 Web Radio"

void mdns_task(EventGroupHandle_t wifi_event_group)
{
    //mdns_server_t * mdns = NULL;

    mdns_result_t * results = NULL;

    while(1) {
        /* Wait for the callback to set the CONNECTED_BIT in the
           event group.
        */
        xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT,
                            false, true, portMAX_DELAY);

        if (!results) {
            ESP_LOGI(TAG, "starting mDNS");

    esp_err_t err = mdns_init();
    if (err) {
        ESP_LOGE(TAG, "Failed starting MDNS: %u", err);
        continue;
    }

            ESP_ERROR_CHECK( mdns_hostname_set(MDNS_HOSTNAME) );
            ESP_ERROR_CHECK( mdns_instance_name_set(MDNS_INSTANCE) );

            ESP_ERROR_CHECK( mdns_service_add(NULL, "_http", "_tcp", 80, NULL, 0) );
            ESP_ERROR_CHECK( mdns_service_instance_name_set("_http", "_tcp", "ESP32 WebServer") );
        }

        // sleep 10 seconds
        vTaskDelay(10000 / portTICK_PERIOD_MS);
    }
}
`

MrBuddyCasino, 2018-01-30:
I've cleaned it up.

closed 2018-01-30 by MrBuddyCasino