MrBuddyCasino/ESP32_Alexa #16: Amazon music support

jhpark555, 2018-02-07:
Is there any ways that I can get amazon music streaming service with this software?
and can I make this device as my device?

MrBuddyCasino, 2018-02-07:
Unimplemented. What do you mean „ my device“?

jhpark555, 2018-02-07:
I thought the token given by activating mac address is not related with my amazon account so I couldn't use the music service. Is there any other reasons not implemented music service?

jhpark555, 2018-02-21:
I have a question.  Once I have a refresh token , why do I have to need refresh token again whenever I do power on.   With my cliendID and client secret , can I get refresh token?

Thanks.

mano1979, 2018-02-22:
@jhpark555 You only need to enter the refresh token once, during "make menuconfig". After programming the esp32, there is no need for tokens anymore (or atleast not from the user). It all goes automaticly during startup.

What i am wondering about is the mac address registration script on Boekling.net; Will it be available indefinetly? Can we get and run that script ourselves?

I designed hardware based on this code, but can we still use it if (hypotheticly) @MrBuddyCasino takes his website down?

MrBuddyCasino, 2018-02-22:
Its a very small Rust binary. I'll publish the code if there is interest.

jhpark555, 2018-02-22:
In the auth_handler.c, below auth_token_refresh() is calling on every power on. Is it normal?

 void auth_token_refresh(alexa_session_t *alexa_session)

and I found there is another access_token routine.   I may not clearly know the process.  Do I need  the refresh token every power on?  I just running the the AVS-DEVICE-SDK on raspberrypi, but I remember I needed only one time refresh token.   Thanks.

mano1979, 2018-02-22:
@MrBuddyCasino Yes i would be interested in the binary.

MrBuddyCasino, 2018-02-24:
I've just created a repository for the authentication backend: https://github.com/MrBuddyCasino/alexa-auth

mano1979, 2018-02-25:
Great! Thank you. I will fork it right away.