MrBuddyCasino/ESP32_Alexa #17: Sorry, but again missing brssl.h

walt22, 2018-02-21:
Hello,

Many thanks for your splendid development!

But unfortunately I still run into the problem with the missing brssl.h.
I did the following:

walter@ubuntu:~/MCProj_ESP32/2018/Alex-in-ESP32/ESP32_Alexa-master$ git init
Initialized empty Git repository in /home/walter/MCProj_ESP32/2018/Alex-in-ESP32/ESP32_Alexa-master/.git/
walter@ubuntu:~/MCProj_ESP32/2018/Alex-in-ESP32/ESP32_Alexa-master$ git submodule init && git submodule update
walter@ubuntu:~/MCProj_ESP32/2018/Alex-in-ESP32/ESP32_Alexa-master$ make
CC asio_secure_socket.o
/home/walter/MCProj_ESP32/2018/Alex-in-ESP32/ESP32_Alexa-master/components/asio/./asio_secure_socket.c:49:19: **fatal error: brssl.h:** No such file or directory
compilation terminated.
/home/walter/esp/esp-idf/make/component_wrapper.mk:210: recipe for target 'asio_secure_socket.o' failed
make[1]: *** [asio_secure_socket.o] Error 1
/home/walter/esp/esp-idf/make/project.mk:390: recipe for target 'asio-build' failed
make: *** [asio-build] Error 2
walter@ubuntu:~/MCProj_ESP32/2018/Alex-in-ESP32/ESP32_Alexa-master$ 

My ../components/bear_ssl folder is empty, but in the folder ../bear_ssl_component the file
component.mk is available with the following content:

COMPONENT_ADD_INCLUDEDIRS := ../bear_ssl/.	\
							../bear_ssl/inc \
							../bear_ssl/src \
							../bear_ssl/tools
							
COMPONENT_SRCDIRS +=	../bear_ssl/src/codec \
						../bear_ssl/src/ec \
						../bear_ssl/src/hash \
						../bear_ssl/src/int \
						../bear_ssl/src/mac \
						../bear_ssl/src/rand \
						../bear_ssl/src/rsa \
						../bear_ssl/src/ssl \
						../bear_ssl/src/symcipher \
						../bear_ssl/src/x509 \
						../bear_ssl/src 
CFLAGS += -DBR_USE_ALT_RAND -DBR_USE_UNIX_TIME

Any idea what went wrong?

Many thanks for any hint
Walter

MrBuddyCasino, 2018-02-21:
I'm afraid not, imho `git submodule init && git submodule update` should fetch the bearssl repo content. Another guy on windows had a similar issue. Not sure what to do.

walt22, 2018-02-22:
Many thanks for your quick reply.
I will not give up. Maybe I try it in an alternative environment, e.g. Debian.
What platform do you use? A virtual machine like me or do you prefer a single installation?
Best regards
Walter

walt22, 2018-02-22:
Same result on my Debian 8 VM.

MrBuddyCasino, 2018-02-22:
I just tried again from a fresh checkout:
```
git clone https://github.com/MrBuddyCasino/ESP32_Alexa.git
cd ESP32_Alexa/
git submodule init && git submodule update
make -j4 flash
```
Works fine, menuconfig opens and then it happily compiles.

walt22, 2018-02-22:
Hurray!!!
It works now with the a.m. instructions. Compiles clean, flashes clean and prints out the MAC address.
I have tested it under Debian 8.
Many thanks for your great help!
Now I will start with the hardware.

Best regards
Walter

closed 2018-02-22 by walt22