MrBuddyCasino/ESP32_Alexa #19: External RAM

adimnno, 2018-03-05:
It seems that you have used the PSRAM module as FIFO for the mic input and mp3 output. 
But you mentioned that you do not need any extra RAM in
[https://www.esp32.com/viewtopic.php?f=17&t=2618&sid=9578357c0bf2c1a5cee7ec16ec83c1f8](url).
How do you achieve playing long response for paragraphs then?

MrBuddyCasino, 2018-03-05:
No, PSRAM is not used. Audio is streaming.

adimnno, 2018-03-05:
Hi MrBuddyCasion,

Thank you very much for your prompt reply.
But I have investigated your code few months before, it seems that you have use the SPI RAM to store the MP3 data replied form AVS before the media player starting to play. It's reasonable as you could not foresee the length of the alexa response.
If you use internal RAM as buffer, ( I guess the RAM size is a great limitation), how could you make sure the smoothness of the playback?

I appreciate much of your great work!!

Thank you very much.

MrBuddyCasino, 2018-03-05:
I guess you mean the SPI Fifo code parts. They are remnants of the ESP8266 origins of this projects, where MP3 playback could only be accomplished with an external SPI RAM chip. The FIFO buffer is still used, but the data does not reside in external RAM anymore. 

A small buffer is used for streaming the audio in real time, and it only works smoothly if your wifi connection is good.

adimnno, 2018-03-05:
OK! Thanks for your pointing out of my question. I would investigate more on your latest code first.

THANK YOU VERY MUCH !!!!

closed 2018-03-05 by MrBuddyCasino