MrBuddyCasino/ESP32_Alexa #21: Wake word support

jonsmirl, 2018-03-20:
Espressif has a wake word engine that runs on the ESP32 now. It works with Alexa.

But... it needs that module with 4MB of PS RAM.

hchaudhary1, 2018-03-20:
open source?

jonsmirl, 2018-03-20:
I don't know...  this is from Espressif

Peter Wang | Peter Wang | 12:33 PM (10 hours ago) |   |  

We have got the IA610 EV board, will make a prototype board for the solution.
If SPH0645LM4H-B be used for near field application ,  I think it is ok.

The wake word can be done on ESP32 or External DSP.  The AEC function should be done on external DSP , if use a Microphone array.
If it is done on ESP32 , then main CPU of ESP32 should keep working to detect and process the wake word.
If it is done on external DSP, then ESP32 can enter sleep mode to saving the power.
That would be useful for battery power device.

jonsmirl, 2018-03-20:
It is part of this kit. But that $10 DSP is not required.

https://www.cnx-software.com/2018/01/16/espressif-esp32-lyratd-ms1-hdk-is-designed-for-smart-speakers-wireless-audio-and-other-smart-home-appliances/

jonsmirl, 2018-03-20:
It need the module with the 4MB PSRAM - ESP32-WROVER
https://www.espressif.com/sites/default/files/documentation/esp32-wrover_datasheet_en.pdf

hchaudhary1, 2018-03-20:
no code released yet... keeping an eye out for it. would be  helpful if they don't need to use the $10 microsemi part

jonsmirl, 2018-03-20:
Peter said...
If SPH0645LM4H-B be used for near field application , I think it is ok.

That is $0.50 part.

The Lryatd kit was released in January.  Ping the CNX guy, he had the kit
Jean-Luc Aufranc <cnxsoft@cnx-software.com>

shirish47, 2018-03-30:
I am also looking for this functionality wakeup word detection. One way I think it can be done is by HM2007 chip(just detecting word and waking ESP32 to send other part of the voice).

yoelrc88, 2018-04-30:
Has someone tried to use the [Cloud-Based Wake Word Verification](https://developer.amazon.com/docs/alexa-voice-service/enable-cloud-based-wake-word-verification.html) to achieve this? I wouldn't mind sending an audio stream to Amazon servers ... at least for testing :)
Thanks

danxster, 2018-06-03:
There's also;
EasyVR Shield 3.0
ELECHOUSE Voice Recognition Module V3
Tigal SmartVR Voice Recognition Board
Grove – Speech Recognizer

jonsmirl, 2018-08-21:
Espressif LyraTD SDK code is here:
https://github.com/espressif/esp-adf

Not sure if the wake word support is in the SDK yet.