MrBuddyCasino/ESP32_Alexa #22: Support for ESP32-LyraT

mlongo95, 2018-05-07:
Just got this amazing dev board and was wondering if this project is compatible with it. I know they both use different audio codecs (ES8388/MAX98357A) so what I'm trying to figure out is how do i change the I2S pin configuration to match the one on the LyraT board. 

Any help is appreciated.

MrBuddyCasino, 2018-05-08:
I don't have that board, so can't really help here.

dhruvchikhaliya12, 2019-12-20:
> Just got this amazing dev board and was wondering if this project is compatible with it. I know they both use different audio codecs (ES8388/MAX98357A) so what I'm trying to figure out is how do i change the I2S pin configuration to match the one on the LyraT board.
> 
> Any help is appreciated.

Would you able to done it with es8388 audio codec i have the same problem.