MrBuddyCasino/ESP32_Alexa #23: Flash Szie

maosuyun2009, 2018-05-09:
When I try to compile this program, 
`ESP32_Alexa/partitions.csv' occupy 2.1MB of flash (2162688 bytes) which does not fit in configured flash size 2MB`
Do i need to use 4M Flash？

MrBuddyCasino, 2018-05-09:
You mean the pure binary size is 2.1MB and your board only has 2MB, or is the CSV partition simply too small and could be enlarged because you physically have 4MB?

hchaudhary1, 2018-05-09:
@maosuyun2009 
You can fit it if you turn off all logging in the menuconfig (bootloader+app), as well as set compiler option of Os

maosuyun2009, 2018-05-10:
@MrBuddyCasino I mean My board only has 2MB,But CSV partition occupy 2.1MB, Can I change CSV partition to fit 2MB Flash

chegewara, 2018-05-26:
This error cames when you have 2MB set flash size in menuconfig. Just change it to 4MB.