MrBuddyCasino/ESP32_Alexa #28: API question

chegewara, 2018-06-02:
Sorry for this simple question, does this library send `SpeachSynthetizer.SpeechFinished` message to amazon servers when esp32 is DONE speaking?

MrBuddyCasino, 2018-06-02:
Not 100% sure, but I think not. The SpeechSynthesizer.SpeechState context with "playerActivity" = "FINISHED" payload is sent when sending speech to the server, but thats it afaik.

chegewara, 2018-06-02:
It is required to play with `Alerts`. 
https://developer.amazon.com/docs/alexa-voice-service/alerts-overview.html#scenario1

Thanks for answer and this great library.

MrBuddyCasino, 2018-06-04:
Looking forward to it.

chegewara, 2018-06-04:
I cant post changes yet, because its paid job, but i hope i will be allowed to make PR, at least some part of code.