MrBuddyCasino/ESP32_Alexa #30: amazon certificate

chegewara, 2018-06-12:
Where i can setup amazon certificate to get this go away:
```
W (35782) http_client: Failed to verify peer certificate!
W (35784) http_client: verification info:   ! The certificate is not correctly signed by the trusted CA
```

Im getting it when im trying to listen flash news which are in mp3 file.
`
{"directive":{"header":{"namespace":"AudioPlayer","name":"Play","messageId":"b575afb7-58b0-4b1c-9c26-90bf8e0d97e9","dialogRequestId":"1"},"payload":{"audioItem":{"audioItemId":"amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#url#ACRI#DailyBriefingPrompt.a9af62ce-5bcb-4ed9-afdb-381a71f70322:ChannelItem:0:0","stream":{"offsetInMilliseconds":0,"expiryTime":"2018-06-12T14:29:39+0000","url":"https://tinytts-eu-west-1.amazon.com/2/3bc7bf3c-6e08-11e8-bbc5-81ee3a22eaed-9026bf/5/1528870627664/f72984a3d31a1f647c405646e1dc9db635063535d54b58501a53fd198151c4fb/resource.mp3","token":"amzn1.as-ct.v1.Domain:Application:DailyBriefing:TTS#ACRI#url#ACRI#DailyBriefingPrompt.a9af62ce-5bcb-4ed9-afdb-381a71f70322:ChannelIntroduction:0"}},"playBehavior":"REPLACE_ALL"}}}
`

MrBuddyCasino, 2018-06-12:
¯\_(ツ)_/¯

chegewara, 2018-06-12:
Found it:
https://github.com/MrBuddyCasino/ESP32_Alexa/blob/master/components/http/http.c#L61-L72

and some help from esp-idf http example:
https://github.com/espressif/esp-idf/blob/release/v3.0/examples/protocols/https_request/main/https_request_example_main.c

But still strugling with implementing it.

PS dont forget this:
https://github.com/espressif/esp-idf/blob/release/v3.0/examples/protocols/https_request/main/component.mk

chegewara, 2018-06-13:
After few more tests ive found real solution. In normal situation it is not recommended, but we will have to handle many https requests from different hosts and maybe there is better solution (download ca root cert from internet for each request) but at least this is some start:
https://github.com/MrBuddyCasino/ESP32_Alexa/blob/master/components/http/http.c#L99
`MBEDTLS_SSL_VERIFY_NONE`

closed 2021-05-17 by chegewara