MrBuddyCasino/ESP32_Alexa #32: support fo v1053 mp3 encoder

gyroing, 2018-06-14:
is it possible to add support to vs1053 mp encoder ?  like KaraDio32

MrBuddyCasino, 2018-06-14:
Why would want that?

gyroing, 2018-06-14:
for 2 purpose : 1- not having the Dac module and better quality  , as wll as  vs1053  has a licensed mp3 decoding.

MrBuddyCasino, 2018-06-14:
Makes no sense to me.
1. I2S already provides full quality at a cheaper cost than the VS1053
2. MP3 decoders never have required a license
3. MP3 patents have all run out anyway

closed 2018-06-14 by MrBuddyCasino

gyroing, 2018-06-14:
Thanks for reply , and i forget vs1053 included internal mic also , so it is possible to combine audio (in-out),

chegewara, 2018-06-14:
i can say that esp32 simple-alexa board designed by @kodera2t is working pretty awesome with this library and is not very expensive (around $26 or 29)