MrBuddyCasino/ESP32_Alexa #36: Make menuconfig

Dema323, 2018-07-10:
Hi im running into some issues with the make menuconfig part. There is something i am missing but i can't figure out what. When i run the make menuconfig i receive the following error: 
```
Makefile:10: /path/to/esp-idf/make/project.mk: No such file or directory
make: *** No rule to make target '/path/to/esp-idf/make/project.mk'.  Stop.
```
I never used the esp tools before so im not so good with it.
Thank you for your time !

chegewara, 2018-07-10:
Do you have setup IDF_PATH in your environment? Check it with `printenv IDF_PATH`.

Dema323, 2018-07-10:
`maxim@DESKTOP-SBL495D MINGW32 ~/esp-idf/ESP32_Alexa
$ printenv IDF_PATH
/path/to/esp-idf

maxim@DESKTOP-SBL495D MINGW32 ~/esp-idf/ESP32_Alexa
$
`
this is the output i get

chegewara, 2018-07-10:
`/path/to/esp-idf`
This needs to be real path to esp-idf folder on you system. Use:
`export IDF_PATH=...` where ... is real path. On windows it will be something like that:
`/c/msys32/home/maxim/esp-idf` where c is drive letter where you have your msys32 and i am assuming you have msys32 in top level folder.

Dema323, 2018-07-10:
Right, i don't know how i missed that. I was copying without actually reading. Dumb mistake. Thank you for your time, im gonna try it now but i should be able to figure it out now. Thanks again !

chegewara, 2018-07-10:
`make menuconfig` and change port in serial flasher config menu to com5.

Dema323, 2018-07-10:
so /dev/ttyusbCOM5 ?

chegewara, 2018-07-10:
its for linux, on windows you have to use only `com5`

Dema323, 2018-07-10:
ah that explains my google search results. Thanks i will try it

Dema323, 2018-07-10:
im sorry to keep bothering, but im getting the following error 

```
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM5': WindowsError(5, 'Access is denied.')
make: *** [/c/users/maxim/desktop/msys32/home/maxim/esp-idf/components/esptool_py/Makefile.projbuild:55: flash] Error 1
```

Dema323, 2018-07-10:
im sorry my atom editor was holding the port open, when i closed it the port was open for the program to execute.

closed 2018-07-17 by Dema323