MrBuddyCasino/ESP32_Alexa #37: suspended program

chen-zifeng, 2018-09-05:
I replaced the https://alexa.boeckling.net in the source code with my https://www.loveandcourage.xyz, compiled the run found code to pause, the log below:

I (1705) wifi: received wifi event 4
I (2725) event: sta ip: 192.168.1.102, mask: 255.255.255.0, gw: 192.168.1.1
I (2725) wifi: received wifi event 7
I (2735) sntp: Initializing SNTP
I (2735) sntp: Waiting for system time to be set... (1/40)
I (2935) sntp: Waiting for system time to be set... (2/40)
I (3135) nghttp2: new nghttp session, uri: https://www.loveandcourage.xyz/auth/refresh/Atzr|IwEBIBJ35jVK1KPGdDtpVR8WfKAWK3Xa0Lr8p4lvNxA5MD2oYlX_6wF2jkbTuZ56dvFgrzBzm-4Am6GVNCMiT4ivU8ED7DmbNuH-gS0DW9WJLON_iYyE4fPwNzU-YYcMZ9GcoL5VKtl7CSNr4VKlsbAb5v3OxRW2UJLMf9jI2TacgHi0I5jWtxGosSG597T4MrriG8lB2A9rjSqx8hcKSyOnKtrpKgIVpmF4sHrni_i9bDxb5t9UZlXVSpQJz74NJzSYpdyduvjka6MqvPRLTqUXVi8d3bMgcsPkYzHtof3e83o7ROQtQZtahPyFT37RIp_yJwvwVe0sGKijDCP8zy-4EXOC8FOV8rhzubyeFKGeU7Yqo_TSpYLXy9S1Iv3mPXMdMxq2jKlOqfFCaUMYEgROfrVNT-CH7EOG4bgYCJN7IbdnAOTz7nphAQLOwblgiGKkwksbCVE9YMLokpBlYCBy6oajQzDELDEbXV4LiEtpi2MG4iQtZawuyiAQXQMCGZRHpP0jn8l4whFcz-JtoqxPTjWKt6dEHmJvrg0Mn_oqzpt54Q
I (3175) nghttp2: Writing HTTP request...
I (3185) nghttp2: Request headers:
I (3185) nghttp2: :method: GET
I (3195) nghttp2: :scheme: https
I (3195) nghttp2: :authority: www.loveandcourage.xyz
I (3195) nghttp2: :path: /auth/refresh/Atzr|IwEBIBJ35jVK1KPGdDtpVR8WfKAWK3Xa0Lr8p4lvNxA5MD2oYlX_6wF2jkbTuZ56dvFgrzBzm-4Am6GVNCMiT4ivU8ED7DmbNuH-gS0DW9WJLON_iYyE4fPwNzU-YYcMZ9GcoL5VKtl7CSNr4VKlsbAb5v3OxRW2UJLMf9jI2TacgHi0I5jWtxGosSG597T4MrriG8lB2A9rjSqx8hcKSyOnKtrpKgIVpmF4sHrni_i9bDxb5t9UZlXVSpQJz74NJzSYpdyduvjka6MqvPRLTqUXVi8d3bMgcsPkYzHtof3e83o7ROQtQZtahPyFT37RIp_yJwvwVe0sGKijDCP8zy-4EXOC8FOV8rhzubyeFKGeU7Yqo_TSpYLXy9S1Iv3mPXMdMxq2jKlOqfFCaUMYEgROfrVNT-CH7EOG4bgYCJN7IbdnAOTz7nphAQLOwblgiGKkwksbCVE9YMLokpBlYCBy6oajQzDELDEbXV4LiEtpi2MG4iQtZawuyiAQXQMCGZRHpP0jn8l4whFcz-JtoqxPTjWKt6dEHmJvrg0Mn_oqzpt54Q

I (3255) nghttp2: created new stream: 1
W (3255) nghttp2: 859: - RAM left 136828
I (3265) asio: adding task: www.loveandcourage.xyz
Algorithms:
I (3265) asio: removing task: refresh_auth_token
W (3275) asio: 96: - RAM left 129252
connecting to: 203.195.166.179
connected.
WARNING: no configured trust anchor
W (3325) led_ui: switching anim
E (3325) led_ui: setColors 0=-2146526380, 1=1073548960
Handshake completed
   version:               TLS 1.2
   cipher suite:          ECDHE_RSA_WITH_AES_256_GCM_SHA384
   ECDHE curve:           secp256r1 (P-256)
   secure renegotiation:  yes
E (3955) asio_http2: closing stream
I (3955) asio_handler_ssl: asio_ssl_handle_close
I (3955) nghttp2: freeing http2_session_data
W (3955) nghttp2: 194: - RAM left 155952
W (3965) asio: 96: - RAM left 156012

Does any god know about this problem？

chegewara, 2018-09-05:
Agree, access token is returned from server and this log looks fine.

chen-zifeng, 2018-09-06:
This is the result of an abnormal operation,If I restore the server address and fill in the corresponding token, the correct log is as follows:
[image]
log:
I (2736) wifi: received wifi event 4
I (3726) event: sta ip: 192.168.1.102, mask: 255.255.255.0, gw: 192.168.1.1
I (3726) wifi: received wifi event 7
I (3736) sntp: Initializing SNTP
I (3736) sntp: Waiting for system time to be set... (1/40)
I (3936) nghttp2: new nghttp session, uri: https://alexa.boeckling.net/auth/refresh/Atzr|IwEBIHsEIgOpH5YuLi6IJa8VhyhqOKQERLfV56sdLNlxgt3EXCY71nyyR-F_1FHyeSYkDAYw1d-sPzu2EPAe5fobGmyxTFlOWpGNZBuzCLqlcLIX56WNLMFqWgiGlD2ksXYk-lJQZkSxYiaIHY4u-yMFX7MtM111tKV8n0RQO5VBBaAsLIMaGmmnQotmvQyE9VzyyGJT3x5K8VoXY8-m6YD_0ZnEHLbJQE_fm3GDCVoq3CTURss2escaWj33DRIK7RtNPphFygYcbe0DbYNcf5bsTQeAEkWT5gxfieY7wFUvu_iF2g5h2lwP86BEb4uIjMHA30BdTUcCIE4qT0_vCbKQMNQn0JM1Ajm9-30qnguqxE2Ym4vyrxyDGrQ95hD8WzGafU4lLQlOUljBVZzDdPNCvWQb0VX85PNhwfDSTTKqMzZgDWAo9xpdxudpiN89tMzUxEnnMiMvYaOldDQa3oRqs0pGEF5R5MoIkpzshiUzKcJJlqQtw-miHi0n2DbUS8v7gQPCpQw9JWmSkM8xjnpV_75qM3Fa9D1DeOX8gsUy7n3ujw
I (3976) nghttp2: Writing HTTP request...
I (3986) nghttp2: Request headers:
I (3986) nghttp2: :method: GET
I (3996) nghttp2: :scheme: https
I (3996) nghttp2: :authority: alexa.boeckling.net
I (3996) nghttp2: :path: /auth/refresh/Atzr|IwEBIHsEIgOpH5YuLi6IJa8VhyhqOKQERLfV56sdLNlxgt3EXCY71nyyR-F_1FHyeSYkDAYw1d-sPzu2EPAe5fobGmyxTFlOWpGNZBuzCLqlcLIX56WNLMFqWgiGlD2ksXYk-lJQZkSxYiaIHY4u-yMFX7MtM111tKV8n0RQO5VBBaAsLIMaGmmnQotmvQyE9VzyyGJT3x5K8VoXY8-m6YD_0ZnEHLbJQE_fm3GDCVoq3CTURss2escaWj33DRIK7RtNPphFygYcbe0DbYNcf5bsTQeAEkWT5gxfieY7wFUvu_iF2g5h2lwP86BEb4uIjMHA30BdTUcCIE4qT0_vCbKQMNQn0JM1Ajm9-30qnguqxE2Ym4vyrxyDGrQ95hD8WzGafU4lLQlOUljBVZzDdPNCvWQb0VX85PNhwfDSTTKqMzZgDWAo9xpdxudpiN89tMzUxEnnMiMvYaOldDQa3oRqs0pGEF5R5MoIkpzshiUzKcJJlqQtw-miHi0n2DbUS8v7gQPCpQw9JWmSkM8xjnpV_75qM3Fa9D1DeOX8gsUy7n3ujw

I (4056) nghttp2: created new stream: 1
W (4056) nghttp2: 859: - RAM left 137316
I (4066) asio: adding task: alexa.boeckling.net
Algorithms:
I (4066) asio: removing task: refresh_auth_token
W (4076) asio: 96: - RAM left 129756
connecting to: 188.166.102.176
connected.
WARNING: no configured trust anchor
W (5136) led_ui: switching anim
E (5136) led_ui: setColors 0=-2146526376, 1=1073548960
Handshake completed
   version:               TLS 1.2
   cipher suite:          ECDHE_RSA_WITH_AES_256_GCM_SHA384
   ECDHE curve:           secp384r1 (P-384)
   secure renegotiation:  yes
   protocol name (ALPN):  h2
I (7446) nghttp2: frame received: 4
I (7446) nghttp2: frame received: 8
I (7806) nghttp2: frame received: 4
I (8316) nghttp2: Response headers for stream ID=1:
I (8316) nghttp2: :status: 200
I (8316) nghttp2: server: nginx/1.10.3 (Ubuntu)
I (8316) nghttp2: date: Thu, 06 Sep 2018 01:40:17 GMT
I (8326) nghttp2: content-type: text/plain; charset=utf-8
I (8326) nghttp2: content-length: 1233
I (8336) nghttp2: strict-transport-security: max-age=63072000; includeSubdomains
I (8336) nghttp2: x-frame-options: DENY
I (8346) nghttp2: x-content-type-options: nosniff

I (8346) nghttp2: All headers received
I (8356) nghttp2: frame received: 0
I (8356) alexa: new auth_token: Atza|IwEBIDc9_mNmj40xrAMflaZNSH-StZu2YjVdrDr2CK2Q6DA2Vbq_mCH_stCgl0vNFFkEPSLj99WAkAZdi597BDsbt-ZivOSDX_m6uGnvdWJCzf2S6UCv2h54blcT-5_3Qs_PNRsRAYNeq7uJzlduHDHwoYcYqCUTTM5lDr68PfAKcbEh0XX9ISfrVDNBNIcX_AjYfB44rZ0pWP8zVUFqwYH3LXbKbkH8XvpjmKR7xetLlFuFhOaoFaHufR40lDBXt2ngkuGyUGhK8gR-wT51u3Wf-pgHPbNZuxvE-QoJQDZPFy4fyA2Qn9jxMYDsqxPGeExF8O3W7M2Kg9Mt-G-ewNnIQClqELN8PcWRcN0zMzbs2Rn2t-DYsUWhwTLSaaeJEElSD5QpzHmKzPfDlHlPeSj5-Id61Lpzybo1N9ji10GfwEPNaXWMKP2V-zzDPI5l5zC5ImSbGXLcbytqh8ljUFYjkelPtj29X-I0x-HRkx7dubHH3gkm35iK4GStxYpRLwGY9LyvOuVBbmUzHbBt0GdPubtLL6LfM6oxNIuRymDId2lBMKTtKHBXrwm3-GfZdLBd2i8
I (8416) asio_http2: closed stream 1 with error_code=0
E (8416) asio_http2: no more open streams, terminating session
E (8426) asio_http2: closing stream
I (8436) asio_handler_ssl: asio_ssl_handle_close
I (8436) nghttp2: freeing http2_session_data
W (8446) nghttp2: 194: - RAM left 156116
W (8446) asio: 96: - RAM left 156180
I (8456) nghttp2: new nghttp session, uri: https://avs-alexa-na.amazon.com/v20160207/directives
I (8456) nghttp2: Writing HTTP request...
I (8466) nghttp2: Request headers:
I (8466) nghttp2: :method: GET
I (8476) nghttp2: :scheme: https
I (8476) nghttp2: :authority: avs-alexa-na.amazon.com
I (8486) nghttp2: :path: /v20160207/directives
I (8486) nghttp2: authorization: Bearer Atza|IwEBIDc9_mNmj40xrAMflaZNSH-StZu2YjVdrDr2CK2Q6DA2Vbq_mCH_stCgl0vNFFkEPSLj99WAkAZdi597BDsbt-ZivOSDX_m6uGnvdWJCzf2S6UCv2h54blcT-5_3Qs_PNRsRAYNeq7uJzlduHDHwoYcYqCUTTM5lDr68PfAKcbEh0XX9ISfrVDNBNIcX_AjYfB44rZ0pWP8zVUFqwYH3LXbKbkH8XvpjmKR7xetLlFuFhOaoFaHufR40lDBXt2ngkuGyUGhK8gR-wT51u3Wf-pgHPbNZuxvE-QoJQDZPFy4fyA2Qn9jxMYDsqxPGeExF8O3W7M2Kg9Mt-G-ewNnIQClqELN8PcWRcN0zMzbs2Rn2t-DYsUWhwTLSaaeJEElSD5QpzHmKzPfDlHlPeSj5-Id61Lpzybo1N9ji10GfwEPNaXWMKP2V-zzDPI5l5zC5ImSbGXLcbytqh8ljUFYjkelPtj29X-I0x-HRkx7dubHH3gkm35iK4GStxYpRLwGY9LyvOuVBbmUzHbBt0GdPubtLL6LfM6oxNIuRymDId2lBMKTtKHBXrwm3-GfZdLBd2i8

I (8546) nghttp2: created new stream: 1
W (8546) nghttp2: 859: - RAM left 134772
I (8556) asio: adding task: avs-alexa-na.amazon.com
Algorithms:
I (8556) asio: removing task: downchannel
[... 63 lines cut ...]
--nghttp2123456789
Content-Disposition: form-data; name="metadata"
Content-Type: application/json; charset=UTF-8

{"event":{"header":{"namespace":"Settings","name":"SettingsUpdated","messageId":"2"},"payload":{"settings":[{"key":"locale","value":"en-US"}]}}}
[... cut ...]
I (15296) nghttp2: Response headers for stream ID=3:
:status: 204
access-control-allow-origin: *
x-amzn-requestid: 0e489ffffe291426-0000070b-00484f84-235909177b2894ae-830eeab5-3

I (15326) alexa: All headers received for stream 3
I (15326) alexa: closed stream 3 with error_code=0
I (15336) asio_http2: closed stream 3 with error_code=0

My purpose is to promote this project to let more people know, I think it is a meaningful code, but I can't move forward on this error problem, I hope you can help me

chen-zifeng, 2018-09-06:
I found different places from logs, but have not found a solution yet
[image]
[image]

chegewara, 2018-09-06:
You need to prepare your server to handle http2 requests (h2) with ECDHE curve P-384 or change alexa code to use for example esp_http_client for this request.

chen-zifeng, 2018-09-07:
after server config http2, system restart
[image]

Can you provide the server configuration for nginx? For example, nginx.config and some other configurations

chegewara, 2018-09-07:
I am using just a bit modified code from this repo:
https://github.com/alexa/alexa-avs-sample-app/tree/master/samples/companionService

chegewara, 2018-09-07:
@chen-zifeng Now you get status 502 html error, maybe bad gateway or something with text/html response which cant be parsed, thats why your app crash. Parser expects text/json response with response code 200.

chen-zifeng, 2018-09-10:
@chegewara Dear friend, I have solved the server problem, but when I look at the recording in the background of alexa, I find that the voice speed is very fast. Do you have this problem? I use a microphone is https://www.adafruit.com/product/3421

chegewara, 2018-09-10:
@chen-zifeng Yes, i also got the same problem, but tweaking some settings solved issue. Dont ask me, i dont remember which one (there is 2 or 3 options you have to check, i also got different board)