MrBuddyCasino/ESP32_Alexa #40: non playable url

jhpark555, 2019-04-12:
From the audio play directive, some url can be playable and some are not.
How can we change the url as playable ?

handle_directive:                                              
{"directive":{"header":{"namespace":"AudioPlayer","name":"Play","messageId":"06f886e4-aaba-4cc9-b211-b01f478a5e9d"},"payload":{"audioItem":{"audioItemId":"amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#35714fe5-f8db-4204-8e6b-2d30b7ea0de5:1","stream":{"offsetInMilliseconds":0,"expiryTime":"2019-04-12T18:29:47+0000","progressReport":{"progressReportDelayInMilliseconds":1000},"url":"https://d29r7idq0wxsiz.cloudfront.net/DigitalMusicDeliveryService/HPS.m3u8?m=m&dmid=200000299180242&c=cf&f=ts&t=10&bl=256k&s=true&e1=1555094700000&e2=1555094700000&v=V2&n=QZDMHAEPXVPVI&h=c6258a02079ff653280d03975e73feedc5bbe4096d61d780b3be57481790b5ba","token":"amzn1.as-ct.v1.Dee-Domain-Music#ACRI#url#ACRI#35714fe5-f8db-4204-8e6b-2d30b7ea0de5:1"}},"playBehavior":"REPLACE_ALL"}}}

playing url https://d29r7idq0wxsiz.cloudfront.net/DigitalMusicDeliveryService/HPS.m3u8?m=m&dmid=200000299180242&c=cf&f=ts&t=10&bl=256k&s=true&e1=1555094700000&e2=1555094700000&v=V2&n=QZDMHAEPXVPVI&h=c6258a02079ff653280d03975e73feedc5bbe4096d61d780b3be57481790b5ba********************handle_play_directive[69]*****************
web_radio_start 134
on_partSeedi the randomng the ran number ge******on_mul************tipart_dat*******on_multipart_data_begin[173]*****************
frame received: 0                                  
Setting hostname for TLS session...
Setting up the SSL/TLS structure...
Connecting to d29r7idq0wxsiz.cloudfront.net:443...
Connected.                                        
Performing the SSL/TLS handshake...
Verifying peer X.509 certificate...
Failed to verify peer certificate! 
verification info:   ! The certificate is not correctly signed by the trusted CA
  
Writing HTTP request...
268 bytes written      
Reading HTTP response...
http_client_get completed
http_client_get stack: 24378

chegewara, 2019-04-13:
m3u8 is playlist text file. You have to read it and retrieve real url from it.

jhpark555, 2019-04-13:
Even though i read the playlist and try with new url, i can’t get to the address either. Below is the example of that. 

Writing HTTP request...
268 bytes written  
Reading HTTP response...
HTTP/1.1 200 OK  
Content-Type: application/vnd.apple.mpegurl
Content-Length: 309
Connection: close
Server: Apache-Coyote/1.1
dmdsRequestId: 8afca961505f3824b96a1d5f03248138
Cache-Control: max-age=1299, s-maxage=0
Content-Disposition: attachment; filename="269785c6-c143-3a81-8501-2f4a7276d5e7.m3u8"
Date: Sat, 13 Apr 2019 00:53:19 GMT
X-Cache: Miss from cloudfront
Via: 1.1 4a55d86b7263f73c6817c7c25d4b3643.cloudfront.net (CloudFront)
X-Amz-Cf-Id: EvXG9He1VJ2lTslMJ4KjBsOoKdB83lutequ166PcZoFBgP6J_JURLg==
 
#EXTM3U
#EXT-X-STREAM-INF:PROGRAM-ID=1,BANDWIDTH=295155,CODECS="mp4a.40.2"
https://d29r7idq0wxsiz.cloudfront.net/DigitalMusicDeliveryService/HPS.m3u8?m=b&dmid=200000309367177&c=cf&f=ts&t=10&b=256k&s=true&e2=1555118100000&v=V2&n=CGX3G3D2FSQU4&h=ef8efdfa3a76ac4fd0c57aab1998b109ca3866fd8bf51995aaf3424d6e26cd80

chegewara, 2019-04-13:
`application/vnd.apple.mpegurl`
its still address of url to mpeg returned, not address of mpeg:
`HPS.m3u8`

jhpark555, 2019-04-14:
> `application/vnd.apple.mpegurl`
> its still address of url to mpeg returned, not address of mpeg:
> `HPS.m3u8`

Do  I need to read attached file  filename="269785c6-c143-3a81-8501-2f4a7276d5e7.m3u8" ?
Thanks.

jhpark555, 2019-04-17:
Finally I got the playlist from the mpegurl. still I have a problem that the http parser values are shifted a little  so I can't get exact value of playlist url.
Thanks.